MarlinFirmware/Marlin #27065: [BUG] Crash when FT_MOTION is turned ON

vovodroid, 2024-05-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I build Marlin with both INPUT_SHAPING_XY and FT_MOTION enabled.  INPUT_SHAPING_XY  works, but when it's disabled, and any FT_MOTION algorithm is enabled X/Y axes just don't move.

With DISTINCT_E_FACTORS and this fix https://github.com/MarlinFirmware/Marlin/pull/27063 Marlin crashes (restarts) after very small axis jerk.

### Expected behavior

FT_MOTION works.

### Actual behavior

Axes don't move or Marlin crashes.

### Steps to Reproduce

Build with both INPUT_SHAPING_XY and FT_MOTION enabled, disable INPUT_SHAPING_XY , enable FT_MOTION, try to move X or Y axis.

### Version of Marlin Firmware

bugfix-2.1.x

### Electronics

SKR-3, TMC2209

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/15252820/Marlin.zip)

vovodroid, 2024-05-09:
Looks like **[BUG] FT_MOTION Homing and Moving Axes #26683**

vovodroid, 2024-05-09:
I tried with
```

#define FTM_STEPPER_FS          100000
#define FTM_STEPPERCMD_BUFF_SIZE 30000
#define MINIMUM_STEPPER_POST_DIR_DELAY 200
```
axes still don't move with FT enabled.

thisiskeithb, 2024-05-09:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/26683

closed 2024-05-09 by thisiskeithb