MarlinFirmware/Marlin #27066: [BUG] 2.1.2.2 severely impacts reliability of BLtouch sensor on Mega2560 8-Bit boards

kb-elmo, 2024-05-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

For some reason after updating to 2.1.2.2 the accuracy of my BLtouch went completely down the drain to the point where it's not usable anymore.
Zero changes to the hardware and no relevant changes to either configuration.h or configration_adv.h except for those minor things required for the update.

On 2.1.2.1 the repeatability test gives me a standard deviation of around 0.002.
On 2.1.2.2 the same BLtouch sensor fails the test with a deviation of sometimes over 0.2mm.

A bed mesh generated on 2.1.2.1 looks like this:
[image]

The mesh on the same machine with zero hardware changes on 2.1.2.2 resulted in this monstrosity:
[image]

It seems that something in the code changed that causes the firmware to not read the reported values of the BLtouch correctly anymore or making the sensor output overly noisy.

Rolling back to 2.1.2.1 fixed the issue so it's definitely firmware related and not caused by a faulty BLtouch or some other hardware related issue.

### Bug Timeline

started after updating to 2.1.2.2

### Expected behavior

The BLtouch should give reliable measurements

### Actual behavior

Unreliable measurements and high std deviation in the repeatability test

### Version of Marlin Firmware

2.1.2.2

### Printer model

Heavily modded Anycubic i3 Mega-S

### Electronics

Anycubic Trigorilla 14 (Ramps 1.4) Mega2560 based 8-bit board

### LCD/Controller

FYSETC_MINI_12864_2_1

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[bltouch_issue_2122_conf.zip](https://github.com/MarlinFirmware/Marlin/files/15253135/bltouch_issue_2122_conf.zip)

thisiskeithb, 2024-05-08:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

noadgskill, 2024-05-10:
In version 2.1.2.2, there are occasional issues during bed leveling where it skips probing points or crashes into the bed. These issues were not present in versions 2.1.2.1 and 2.0.9.7. The probe I use is an IR sensor, and there were no problems before the update.

thisiskeithb, 2024-05-10:
> In version 2.1.2.2, there are occasional issues during bed leveling where it skips probing points or crashes into the bed. These issues were not present in versions 2.1.2.1 and 2.0.9.7. The probe I use is an IR sensor, and there were no problems before the update.

As I mentioned above, please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

noadgskill, 2024-05-10:
> As I mentioned above, please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this [...]

Yes, I tested with the latest version today, but the same issue occurs. The control board I am using is the FYSETC E4 V1.0.

kb-elmo, 2024-05-10:
I'll do it once I have my current project finished and have time to flash the new firmware on it again.
Unfortunately I only have this one printer and I need it functioning right now 😄 

I already tested the bugfix branch btw, same result.
Highly unrealiable BLtouch measurements and unusable bed mesh.

markotime, 2024-05-17:
My problem is a failure to probe at the first or a subsequent location after the center-table probe.  Z seems to rise a few mm before descending, and never quite getting to the table surface.  MKS GEN L V1

markotime, 2024-05-17:
Lots of compile errors around probe type options, "#IF" statements following disabled declarations

closed 2024-08-30 by github-actions[bot]