MarlinFirmware/Marlin #27076: [BUG] No axes move with FT_MOTION

vovodroid, 2024-05-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

No XY axes move with merged  **FT_MOTION : updating last_direction_bits for Core printer solves homing failure #26720** 

When FT_MOTION is enabled in menu no XY homing axis move happens.

### Version of Marlin Firmware

[cron] Bump distribution date (2024-05-10)

### Printer model

Cartesian

### Electronics

SKR-3, TMC2209

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/15271902/Marlin.zip)

thisiskeithb, 2024-05-10:
Sounds like @narno2202 will be unable to help out for the foreseeable future, but @ulendoalex may be able to help troubleshoot.

danbrianwhite, 2024-05-11:
I'm having a similar with the recent FT_MOTION PR merge. I am unable to home my core_xy printer with FT_MOTION enabled for build. If I disable it fully from build and comment out the feature, I can home and print properly.

I tried, with FT_MOTION defined for build. I tried both endstop and sensorless homing. I also tried with and without enabled by default without success.

When enabled/on, the core_xy printer homes moving the print head to the diagonal corner instead.of x/y only. It does not detect the endstop. When ft_motion used for build but disabled for homing, it moves properly straight to the endstop, but it is not triggered when pressed.

Thoughts?

vovodroid, 2024-05-11:
I'm not sure whether it's the same issue or not. I have cartesian printer, and axes just don't start to move. But enabling FTM in build doesn't harm, only turning it on in menu.

narno2202, 2024-05-17:
Hello guys, just few words. When @thinkyhead  add @ulendoalex'scode in #26720, some code were missing, I corrected a few mistakes but it was incomplete. This PR was also buggy on my printer. Could you please give feedback on #26848. It's not the same code for endstops monitoring : last known working code of #26720 and other changes.

vovodroid, 2024-05-17:
> Could you please give feedback on #26848. I

Would you like me to merge your PR to bugfix-2.1.x and check? Or just take your branch?

thisiskeithb, 2024-05-17:
> Would you like me to merge your PR to bugfix-2.1.x and check? Or just take your branch?

thinkyhead has already merged in the latest `bugfix-2.1.x` and is fixing up other things right now.

grizzleeadam, 2024-05-20:
I was just toying around with the latest Bugfix (from 5/19/24), and found that FT Motion seems to be working fine for me - so long as Adaptive_Step_Smoothing and Edge_Stepping are disabled. Either one of those seems to cause inability to move & crashing into the bed on homing. I'm on an SKR E3 v1.2.

dbuezas, 2024-07-06:
I wanted to give FT Motion a try and found (5/July/2024 bugfix 2.1.x):
* FTMotion ON: Steppers make noises skip and generally behave weirdly (I think the acceleration stage is skipped, the steppers wait for a moment and then instantly go to target speed)
* FTMotion ZV: Can't move any axis
* Other input shapers: same, can't move any axis but Z

I tried disabling Adaptive_Step_Smoothing and Edge_Stepping and it made no difference.

(Skr3 with EZ2209)

closed 2024-08-10 by vovodroid