MarlinFirmware/Marlin #27077: [BUG] New encoder logic & debounce (#26723) bricks

classicrocker883, 2024-05-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Upon flashing new firmware, stuck on starting screen.

### Bug Timeline

New encoder logic & debounce (#26723)

### Expected behavior

Expect to start normally, to main menu

### Actual behavior

Flashes, but does not boot past starting menu

### Steps to Reproduce

1. Compile with E3V2 DWIN LCD, (e.g. ProUI)
2. Load on SD, then flash
3. Observe LCD

### Version of Marlin Firmware

bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-05-10:
Please attach your configuration files.

classicrocker883, 2024-05-10:
im going to say false alarm for now. there was something i missed to check. but I can affirm a different type almost similar bug with ProUI

closed 2024-05-10 by classicrocker883