MarlinFirmware/Marlin #27078: Single pin for BTN for lcd connectors [FR] (feature summary)

Reza-v, 2024-05-10:
### Is your feature request related to a problem? Please describe.

So i am designing a control board based on the stm32 blackpill and because of a limited number of pins, i've had an idea. I'm trying to combine the three control wheel pins, BTN_EN1, EN2, and ENC and connect them to a single analog pin through resistors with different resistances and have the controller sense the resistances and activate the buttons accordingly. Now i am not very adept in programming MCUs so any help is appreciated. Should i just write the conditions in the pins file and that would work? I haven't tried anything yet.

### Describe the feature you want

Doodooo deedee daa

ellensp, 2024-05-10:
I dont think this can be done.

the encoder uses quadrature encoding

Ie there are times when both BTN_EN1 and BTN_EN2 are connected

[image]

Reza-v, 2024-05-10:
It's definitely possible through software it just might take up some flash space
And while there are times when both pins are high/low, that equates to another resistance value that can be picked up

van0014, 2024-07-05:
While it would be nice to have it all on one pin, because of the rotary encoder, it's not just a case of using a resistor ladder. 

The rotary encoder should be using an ISR for precise and smooth detection. A resistor ladder would likely be too slow to respond, while it is factoring in multiple different possible values.

Do you have the hardware side of this planned out? It cannot be added to Marlin without having an idea about what the hardware layout is, and whether single wire rotary encoders are even useable. More likely, you'd have to use some 3 wire serial data system in order to use minimal wiring

thinkyhead, 2026-05-18:
Implement something and submit a PR with the working code. If there is a hardware product coming out of the project that needs support in the firmware we will add it as a supported feature.