MarlinFirmware/Marlin #27079: [BUG] MPCTEMP and 12864 LCD gives error: expected unqualified-id before numeric constant

classicrocker883, 2024-05-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

`MPCTEMP` and `CR10_STOCKDISPLAY` give this error

```r
/home/runner/.platformio/packages/framework-arduinoststm32-maple/STM32F1/libraries/Wire/utility/WireBase.h:50:19: error: expected unqualified-id before numeric constant
 #define SUCCESS   0        /* transmission was successful */
                   ^
Marlin/src/lcd/dogm/../../module/temperature.h:1249:54: note: in expansion of macro 'SUCCESS'
           enum MeasurementState { CANCELLED, FAILED, SUCCESS };
                                                      ^~~~~~~
In file included from Marlin/src/lcd/dogm/marlinui_DOGM.cpp:54:0:
Marlin/src/lcd/dogm/../../module/temperature.h:1250:25: error: expected unqualified-id before 'const'
           MPC_autotuner(const uint8_t extruderIdx);
                         ^~~~~
Marlin/src/lcd/dogm/../../module/temperature.h:1250:25: error: expected ')' before 'const'
```

### Bug Timeline

in the past couple weeks

### Expected behavior

to compile correctly

### Actual behavior

gives error at compile

### Steps to Reproduce

1. Use Configs or Ender-3 comparable
2. #define Ender-3/CR10 Stock 12864 LCD 
3. #define `MPCTEMP`

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Voxelab Aquila

### LCD/Controller

12864

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlinconfigs.zip](https://github.com/MarlinFirmware/Marlin/files/15280067/Marlinconfigs.zip)

thisiskeithb, 2024-05-10:
This appears to be a GD/Maple-only issue.

Compiling your config works fine with `STM32F103RE_creality` and all my STM32 HAL-based configs with MPCTEMP and any 12864-based LCD compile fine.

classicrocker883, 2024-05-11:
it worked fine about 2 or 3 weeks ago

classicrocker883, 2024-05-11:
it was `#include <Wire.h>` being in **_Bootscreen.h**, I dont know why that was in there but that was the issue

closed 2024-05-11 by classicrocker883