MarlinFirmware/Marlin #27080: [BUG] Deploy/Stow wrong behaviour with a magnetic mounted probe

Arloag, 2024-05-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

First of all, **thank you very much** to all the developers for the great work!!!!

And now the problem:
I´ve added a magnetic mounted probe on my self-made delta 3D printer, but the behaviour of the deploy and stow movements are not working correctly (in my humble opinion).

After homing, the deploy funtion makes no movement (neither with M401 in repetier nor with the option in the printer menu). 
The stow function makes the correct movements, but after them a Z probe error appears (err: ZPROBE)
  

### Bug Timeline

I also had the same problem with the previous version of 2.1.2.2 (maybe the 2.1.1.?)

### Expected behavior

- The deploy function should move the effector to Z100, then move it 2cm to the left (X-20) and then return to the center (X0).
- The stow function should do the same but in the oposite side: move the effector to Z100, then move it 2cm to the right (X20) and then return to the center (X0). 

These movements are just for testing pouposes.
 
The configuration is:

#define MAG_MOUNTED_PROBE
#if ENABLED(MAG_MOUNTED_PROBE)
  #define PROBE_DEPLOY_FEEDRATE (50*60)  // (mm/min) Probe deploy speed
  #define PROBE_STOW_FEEDRATE   (50*60)  // (mm/min) Probe stow speed

  #define MAG_MOUNTED_DEPLOY_1 { PROBE_DEPLOY_FEEDRATE, { 0, 0, 100 } }  // Move to side Dock & Attach probe
  #define MAG_MOUNTED_DEPLOY_2 { PROBE_DEPLOY_FEEDRATE, { -20, 0, 100 } }  // Move probe off dock
  #define MAG_MOUNTED_DEPLOY_3 { PROBE_DEPLOY_FEEDRATE, { 0, 0, 100 } }  // Extra move if needed
  //#define MAG_MOUNTED_DEPLOY_4 { PROBE_DEPLOY_FEEDRATE, { 0, 0, 100 } }  // Extra move if needed
  //#define MAG_MOUNTED_DEPLOY_5 { PROBE_DEPLOY_FEEDRATE, { 0, 0, 100 } }  // Extra move if needed

  #define MAG_MOUNTED_STOW_1   { PROBE_STOW_FEEDRATE,   { 0, 0, 100 } }  // Move to dock
  #define MAG_MOUNTED_STOW_2   { PROBE_STOW_FEEDRATE,   { 20, 0, 100 } }  // Place probe beside remover
  #define MAG_MOUNTED_STOW_3   { PROBE_STOW_FEEDRATE,   { 0, 0, 100 } }  // Side move to remove probe
  //#define MAG_MOUNTED_STOW_4   { PROBE_STOW_FEEDRATE,   { 0, 0, 100 } }   // Side move to remove probe
  //#define MAG_MOUNTED_STOW_5   { PROBE_STOW_FEEDRATE,   { 0, 0, 100 } }  // Extra move if needed

### Actual behavior

- The deploy function does any movement 
- The stow function does the correct movements, but then a Z probe error appears.

In my opinion, since the deploy and stow functions should just pick up the probe from its support and return it back, the probe itself should perfom any action, therefore I don´t understand why the Z probe error appears (err: ZPROBE). 

### Steps to Reproduce

1. Connect the printer
2. Home the printer
3. Check the deploy and stow functions through the printer´s menu or through repetier (with M401 and M402) 

### Version of Marlin Firmware

2.1.2.2

### Printer model

Delta self-made

### Electronics

BIGTREETECH SKRV1.3, drivers TMC2209

### LCD/Controller

REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

### Other add-ons

magnetic mounted probe

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/15280058/Configuration_adv.zip)
[Log output.txt](https://github.com/MarlinFirmware/Marlin/files/15280107/Log.output.txt)

Arloag, 2024-08-15:
Hi. I will check it again with the last bugfix-2.1.x version, but next week (now I'm away from my printer). 
Thank you very much for your work!!!

Arloag, 2024-09-15:
Hi. I finally managed to test it again with the last bugfix-2.1.x version (donwloaded on 08.Sep) and the problem still persist.

**Expected behavior:**
- The deploy function should move the effector to Z100, then move it 2cm to the left (X-20) and then return to the center (X0).
- The stow function should do the same but in the oposite side: move the effector to Z100, then move it 2cm to the right (X20) and then return to the center (X0).
(These movements are just for testing pouposes.)

**Actual behavior:**
- The deploy function does any movement
- The stow function does the correct movements, but then a Z probe error appears.

Any idea how to solve this problem? I've been struggling with this for over 6 months, and I´m quite frustrated 😭

pmcclay, 2025-02-15:
Same. Trying to set up Ender 3 Pro V4.2.2 w/GD32F303RET6 (maple) & klackender. Supposedly it supposed to work but this is as far as I get.

pmcclay, 2025-02-16:
@Arloag are you testing the stow/deploy actions before connecting your mag switch to the controller board (e.g. in place of an NC Z limit switch)?  If so, trigger/hold the existing input (e.g. Z limit) before invoking the deploy action.
See PROBE_TRIGGERED_WHEN_STOWED_TEST in config.

thinkyhead, 2026-10-08:
Following up for the 2.1.3 release: this one is still waiting on the details requested above. Could you test with the current `bugfix-2.1.x` and post your configs and the results? If we don't hear back in a couple of weeks we'll close it, but it can always be reopened.