MarlinFirmware/Marlin #27082: [BUG] FT Motion causes endstops to not trigger properly

barleysvisuals, 2024-05-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When I enable #define FT_MOTION the printer rams into the endstop (noticed on X axis, don't want to test the others).
The error is reproducable and I narrowed it down to FT Motion.

### Bug Timeline

When trying out the new firmware

### Expected behavior

Triggers work as intended

### Actual behavior

Gantry rams into the endstop.

### Steps to Reproduce

1. Enable #define FT_MOTION in configuration
2. Flash printer with new firmware
3. Home axis

### Version of Marlin Firmware

2.1.x Bugfix Version

### Printer model

Creality Ender 3 Pro

### Electronics

Silent Board 32 Bit (4.2.7)

### Other add-ons

Nothing that should affect this

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_BUG.zip](https://github.com/MarlinFirmware/Marlin/files/15283153/Configuration_BUG.zip)

thisiskeithb, 2024-05-11:
Related issue: https://github.com/MarlinFirmware/Marlin/issues/27076

FT Motion doesn't appear to work correctly right now.

Weresk, 2024-08-11:
Any updates since release bug?

thinkyhead, 2024-08-11:
Patches are pending and we'll be reviewing and testing various changes to FT Motion over the coming weeks to make sure it's ready for use at the time of release. I'll be sure to set up a dedicated channel for this sub-project on Discord as well.

thinkyhead, 2024-12-04:
If this is resolved for you with 2.1.3-b1 then we can close this. Otherwise we'll need to keep looking. AFAIK the only issues with homing are that it's tricky to get sensorless in the right range, and BIQU Microprobe has been choking, for some reason.

barleysvisuals, 2024-12-05:
Hi,
I'm currently not able to test this anymore as i switched to Klipper :0 
But thanks for (probably) fixing this 👍

closed 2024-12-05 by barleysvisuals