MarlinFirmware/Marlin #27091: [BUG] simulator is failing to build on windows 11

Asd677gg, 2024-05-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Simulator for windows 11 is not building

### Bug Timeline

New

### Expected behavior

I expected it to build normally

### Actual behavior

Instead, a lot of errors and warnings occured and the build failed

### Steps to Reproduce

1) download marlin-bugfix-2.1.x and configurations-bugfix-2.1.x
2) change configuration.h and configuration.adv.h to simulator example
3) install msys2 and mingw64 with reqiued packages as described in ini/native.ini
4) set default_envs = simulator_windows in the platformio.ini
5) set BOARD_SIMULATED in configuration.h
6) run simulator_windows environment in auto build marlin

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Simulated

### Electronics

Simulated

### LCD/Controller

Simulated

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/15306344/Config.zip)

<details>
  <summary>Terminal (Logs)</summary>
  
```
  PS C:\Users\user1\Documents\MarlinSimulator\Marlin-bugfix-2.1.x> platformio run --silent -e simulator_windows
[... 415 lines cut ...]
   20 | #       error "GLM: GLM_GTX_compatibility is an experimental extension and may change in the future. Use #define GLM_ENABLE_EXPERIMENTAL before including it, if you really want to use it."
      |         ^~~~~
*** [.pio\build\simulator_windows\debug\liba30\MarlinSimUI\MarlinSimulator\application.o] Error 1
In file included from .pio\libdeps\simulator_windows\MarlinSimUI\src\MarlinSimulator\visualisation.h:18,
                 from .pio\libdeps\simulator_windows\MarlinSimUI\src\MarlinSimulator\application.h:11,
                 from .pio\libdeps\simulator_windows\MarlinSimUI\src\MarlinSimulator\main.cpp:4:
C:/msys64/mingw64/include/glm/gtx/euler_angles.hpp:22:9: error: #error "GLM: GLM_GTX_euler_angles is an experimental extension and may change in the future. Use #define GLM_ENABLE_EXPERIMENTAL before including it, if you really want to use it."
   22 | #       error "GLM: GLM_GTX_euler_angles is an experimental extension and may change in the future. Use #define GLM_ENABLE_EXPERIMENTAL before including it, if you really want to use it."
      |         ^~~~~
In file included from C:/msys64/mingw64/include/glm/gtx/euler_angles.inl:3,
                 from C:/msys64/mingw64/include/glm/gtx/euler_angles.hpp:333:
C:/msys64/mingw64/include/glm/gtx/compatibility.hpp:20:9: error: #error "GLM: GLM_GTX_compatibility is an experimental extension and may change in the future. Use #define GLM_ENABLE_EXPERIMENTAL before including it, if you really want to use it."
   20 | #       error "GLM: GLM_GTX_compatibility is an experimental extension and may change in the future. Use #define GLM_ENABLE_EXPERIMENTAL before including it, if you really want to use it."
      |         ^~~~~
*** [.pio\build\simulator_windows\debug\liba30\MarlinSimUI\MarlinSimulator\main.o] Error 1
================================================================================================= [FAILED] Took 103.49 seconds =================================================================================================

Environment        Status    Duration
-----------------  --------  ------------
simulator_windows  FAILED    00:01:43.488
============================================================================================= 1 failed, 0 succeeded in 00:01:43.488 =============================================================================================
PS C:\Users\user1\Documents\MarlinSimulator\Marlin-bugfix-2.1.x> echo "done" >"C:\Users\user1\AppData\Local\Temp\ipc"
```
  
</details>

ellensp, 2024-05-14:
Huh?

2. "change configuration.h and configuration.adv.h to simulator example" good so far
5. "set BOARD_SIMULATED in configuration.h"  why did you have t do this?, this should already be set in 2.

Asd677gg, 2024-05-14:
Sorry for the error, it's my first time opening a issue and it's my third time compiling a marlin firmware

ellensp, 2024-05-14:
Update nativei.ini with

```
[env:simulator_windows]
extends         = simulator_common
build_src_flags = ${simulator_common.build_src_flags} -fpermissive
build_flags     = ${simulator_common.build_flags} ${simulator_common.debug_build_flags}
                  -IC:\\msys64\\mingw64\\include\\SDL2 -fno-stack-protector -Wl,-subsystem,windows
                  -ldl -lmingw32 -lSDL2main -lSDL2 -lSDL2_net -lopengl32 -lssp
                  -DGLM_ENABLE_EXPERIMENTAL
build_type      = debug

```

sjasonsmith, 2024-05-14:
@ellensp  what did you change? I’ve certainly built the windows simulator very recently. How was it broken?

ellensp, 2024-05-14:
@sjasonsmith new install, new everything 

Exactly as the log says

C:/msys64/mingw64/include/glm/gtx/euler_angles.hpp:22:9: error: #error "GLM: GLM_GTX_euler_angles is an experimental extension and may change in the future. Use #define GLM_ENABLE_EXPERIMENTAL before including it, if you really want to use it."
   22 | #       error "GLM: GLM_GTX_euler_angles is an experimental extension and may change in the future. Use #define GLM_ENABLE_EXPERIMENTAL before including it, if you really want to use it."
      |         ^~~~~
In file included from C:/msys64/mingw64/include/glm/gtx/euler_angles.inl:3,
                 from C:/msys64/mingw64/include/glm/gtx/euler_angles.hpp:333:
C:/msys64/mingw64/include/glm/gtx/compatibility.hpp:20:9: error: #error "GLM: GLM_GTX_compatibility is an experimental extension and may change in the future. Use #define GLM_ENABLE_EXPERIMENTAL before including it, if you really want to use it."
   20 | #       error "GLM: GLM_GTX_compatibility is an experimental extension and may change in the future. Use #define GLM_ENABLE_EXPERIMENTAL before including it, if you really want to use it."
      |         ^~~~~

Asd677gg, 2024-05-14:
> Update native.ini with new env:simulator_windows
> 
> ```
> [env:simulator_windows]
> extends         = simulator_common
> build_src_flags = ${simulator_common.build_src_flags} -fpermissive
> build_flags     = ${simulator_common.build_flags} ${simulator_common.debug_build_flags}
>                   -IC:\\msys64\\mingw64\\include\\SDL2 -fno-stack-protector -Wl,-subsystem,windows
>                   -ldl -lmingw32 -lSDL2main -lSDL2 -lSDL2_net -lopengl32 -lssp
>                   -DGLM_ENABLE_EXPERIMENTAL
> build_type      = debug
> ```

Thanks, it compiles now.

closed 2024-05-14 by Asd677gg