MarlinFirmware/Marlin #27093: [BUG]  error: 'ENCODER_5X_STEPS_PER_SEC' was not declared in this scope in latest compile for Monoprice Maker Select V2

william-orange, 2024-05-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Can not compile for Monoprice Maker Select V@ using latest bugfix.  The error is Marlin\src\lcd\marlinui.cpp:1050:24: error: 'ENCODER_5X_STEPS_PER_SEC' was not declared in this scope

### Bug Timeline

Problem appeared 5/14/2024

### Expected behavior

Expected compile wit no error

### Actual behavior

Compile failed as shown

### Steps to Reproduce

attempt compile

### Version of Marlin Firmware

latest bugfix

### Printer model

Monoprice Maker Select V2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/15310020/Configs.zip)

thisiskeithb, 2024-05-14:
> Can not compile for Monoprice Maker Select V@ using latest bugfix. 

Your configs are not up to date with current [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/refs/heads/bugfix-2.1.x.zip), but they compile fine:

```prolog
Linking .pio/build/sanguino1284p_optimized/firmware.elf
Checking size .pio/build/sanguino1284p_optimized/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  38.8% (used 6364 bytes from 16384 bytes)
Flash: [========= ]  88.7% (used 112594 bytes from 126976 bytes)
Building .pio/build/sanguino1284p_optimized/firmware.hex
================================== [SUCCESS] Took 20.46 seconds ==================================

Environment              Status    Duration
-----------------------  --------  ------------
sanguino1284p_optimized  SUCCESS   00:00:20.462
=================================== 1 succeeded in 00:00:20.462 ===================================
```

closed 2024-05-14 by thisiskeithb

ellensp, 2024-05-14:
this is not possible

1) tried your configs, they build fine
2) ENCODER_5X_STEPS_PER_SEC is  automatically added in code

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/inc/Conditionals_adv.h#L879-L882

```cpp
#if ENABLED(ENCODER_RATE_MULTIPLIER)
  #ifndef ENCODER_5X_STEPS_PER_SEC
    #define ENCODER_5X_STEPS_PER_SEC 0
  #endif
```

and #define ENCODER_RATE_MULTIPLIER is in Your Configuration_adv.h

william-orange, 2024-05-14:
And yet on my end it does not build.  I added #define ENCODER_5X_STEPS_PER_SEC   15 // (steps/s) Encoder rate for 5x speed around line 1486 (before the one for 10X) in my configuration_adv.h and I can now compile.

thisiskeithb, 2024-05-14:
Your copy of Marlin may be broken.

Download a fresh copy of [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip), apply/migrate your config (or use an example config if you didn't change anything) and you'll be set.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server