MarlinFirmware/Marlin #27096: [BUG] M914 extra axis name mismatch

Dakkaron, 2024-05-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When reporting the sensitivity values using `M914`, Marlin uses the names `I`, `J` and so on for extra axis. But when setting these values, I need to use the names `A`, `B` and so on.

### Expected behavior

The names used should be consistent. Both when setting and reporting the values, the axies should be named `A`, `B` and so on.

### Actual behavior

When reporting the axies values, Marlin uses `I`, `J`, ...

### Steps to Reproduce

1. `M914` -> `I homing sensitivity: 200`
2. `M914 A100`
3. `M914` -> `I homing sensitivity: 100`

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

DIY, based on Ender 5, Octopus v1.1, E3D Toolchanger

### Electronics

Octopus v1.1

### LCD/Controller

Stock Ender 5 LCD

### Other add-ons

E3D Toolchanger toolchanger using stepper (axis A) to change tools

### Bed Leveling

No Bed Leveling

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/15324960/Marlin.zip)

Dakkaron, 2024-08-14:
Yes, still relevant.

thisiskeithb, 2024-08-14:
I believe this is intentional:

> `AXISn_NAME` defines the letter used to refer to the axis in (most) G-code commands.

From `Configuration.h`:

https://github.com/MarlinFirmware/Marlin/blob/ed7f3b21c06c5343e48d469dda0580610de2deca/Marlin/Configuration.h#L180-L191

So you must use the the `AXISn_NAME` value from your config to set extra G-code parameters, but Marlin reports the internal name.

ellensp, 2024-08-14:
I'm not sure if its a bad idea or not...  but try this

in Marlin/src/feature/tmc_util.h is 

```cpp
    void printLabel() {
      SERIAL_CHAR(AXIS_LETTER);
      if (DRIVER_ID > '0') SERIAL_CHAR(DRIVER_ID);
    }
```

replace it with

```cpp
    void printLabel() {
      char axis_name = AXIS_LETTER;
      #if HAS_I_AXIS
        if (AXIS_LETTER == 'I') axis_name = AXIS4_NAME;
      #endif
      #if HAS_J_AXIS
        if (AXIS_LETTER == 'J') axis_name = AXIS5_NAME;
      #endif
      #if HAS_K_AXIS
        if (AXIS_LETTER == 'K') axis_name = AXIS6_NAME;
      #endif
      SERIAL_CHAR(axis_name);
      if (DRIVER_ID > '0') SERIAL_CHAR(DRIVER_ID);
    }
```

See if this does what you wnt

closed 2024-11-27 by github-actions[bot]