MarlinFirmware/Marlin #27097: [BUG] (Full Fraphic Smart controller + RAMPS not working LCD menu)

Axis008, 2024-05-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Full Fraphic Smart controller + RAMPS on Arduino Mega after firmware install printer loaded succefully but when press button on encoder button on LCD printer reboot. I realize what bug repeated with LCD_LANGUAGE ru . If chagne language to en - it working fine
Hardware not broken because I just update from version 1.19 - before everything working well

### Bug Timeline

New bug, found 15.04.2024

### Expected behavior

I expect to run menu on Full Fraphic Smart controller + RAMPS on Arduino Mega 

### Actual behavior

When press encoder button on LCD printer reboot

### Steps to Reproduce

1. install marlin on Full Fraphic Smart controller + RAMPS on Arduino Mega  with #define LCD_LANGUAGE ru and #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
2. Press encoder button on LCD display 

### Version of Marlin Firmware

Marlin bugfix-2.1.x (May 15 2024 13:46:12) 

### Printer model

Custom

### Electronics

RAMPS on Arduino Mega

### LCD/Controller

Full Fraphic Smart controller + 

### Bed Leveling

No Bed Leveling

### Your Slicer

None

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Uploading Configuration.zip…]()

ellensp, 2024-05-16:
I can replicate this bug

ellensp, 2024-05-16:
Its apparently been broken for some time

I did a git bisect on this and identified commit https://github.com/MarlinFirmware/Marlin/commit/5523c12cfd01c859a01f26575a1dfff4c43eac8e as the cause of the reboot
ie https://github.com/MarlinFirmware/Marlin/pull/26381

closed 2024-05-19 by thinkyhead