MarlinFirmware/Marlin #27098: MKS WiFi module does not work with MKS Eagle and H43 display

spammchik, 2024-05-17:
### Is your feature request related to a problem? Please describe.

There is MKS WiFi module with MKS Eagle and H43 display.
WiFi in Marlin firmware is configured only to work via TFT_LVGL_UI
It is impossible to run MKS WiFi module on MKS Eagle and H43 display.
The same problem will occur on Robin nano V3 and H43 display.
At the same time, the MKS Eagle and Robin nano V3 boards have a connector for connecting the MKS WiFi module and the H43 display.
However, due to unfinished firmware, this equipment does not work together.
Please add to Marlin the ability to work the MKS WiFi module with the H43 display for MKS Eagle and Robin nano V3.

### Are you looking for hardware support?

MKS WiFi module, H43 display, MKS Eagle, Robin nano V3

### Describe the feature you want

Please add to Marlin the ability to work the MKS WiFi module with the H43 display for MKS Eagle and Robin nano V3.

ellensp, 2024-05-17:
Perhaps  your referring to just this one line in the Config?
 
```
#if ENABLED(TFT_LVGL_UI)
  //#define MKS_WIFI_MODULE // MKS WiFi module
#endif
```

This is trivial to change to  #if DGUS_UI_IS(MKS) || ENABLED(TFT_LVGL_UI,)

spammchik, 2024-05-20:
> This is trivial to change to #if DGUS_UI_IS(MKS) || ENABLED(TFT_LVGL_UI) 

Unfortunately, it just doesn't work that way.
MKS says the following in 2021:
https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/wiki/Use-MKS_Robin_WIFI-on-Robin-Nano
**Currently only LVGL_UI supports MKS Wifi function.**

3DSmitty, 2024-06-26:
So the ESP3D module and the H43 both talk to the mainboard via serial communication port. Wifi is on serial port 1 and H43 is on serial port 3. Set these up in Marlin you should be fine.

thisiskeithb, 2024-06-27:
This would require updates within Marlin & the H43 firmware to add the desired menus/features if `MKS_WIFI_MODULE` is enabled.

Having said that, I just confirmed that an ESP8266/ESP-12S WiFi module flashed with ESP3D & WiFi serial port (`1`) works fine on an MKS Eagle without additional changes. I can access the ESP3D web interface & communicate with the motherboard.