MarlinFirmware/Marlin #27103: [BUG] Mini12864 display panel works until the SD card is accessed

RzTen1, 2024-05-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I've attached a BGTHREETECH MINI 12864 V2.0 display board and knob to my BIGTREETECH SKR MINI E3 V3.0 control board via a SKR Mini Screen Adapter and the display operates normally until the SD card is accessed.  Once the menu to view the card is selected the display stops responding.  The SPI clock also stops being generated.  While this is happening I can issue a M21 command from the serial console and the card is detected and the files are listed, along with the brief reappearance of the SPI clock, but once it's done the display remains stuck.  Removing the SD card causes the LCD to reset and resume normal operation.

I've tried both the controller's onboard SD card slot and the SD card slot on the display and they both work, but they also both cause the display to stall.  The display's SD card slot was enabled by modifying the boards pins to use an extra I/O header for the SD chip select and card detect and setting the chip select line on the onboard SD card to high/disabled. It appears that accessing the SD card causes the display loop to jam for some reason.

If the printer is booted with an SD card inserted the Marlin logo appears and then the display locks up.  I can still talk to the controller via the serial connection.  The 'media inserted' text does display when a card is inserted into a slot and the screen keeps functioning normally, it's just when the directory is listed it stalls.

### Bug Timeline

New bug

### Expected behavior

The display should show a list of files and allow one to be selected.  On boot the normal status screen should appear.

### Actual behavior

The display gets stuck displaying it's last image until the SD card is removed.

### Steps to Reproduce

1. Boot with no SD card inserted.
2. Insert a SD card into either the onboard SD card slot or the display's SD card slot, depending on configuration.
3. Observe that the 'media detected' text appears at the bottom of the status display.
4. Enter the menu and try to list the files on the card via the display and control wheel.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

it's a mix of a Solidoodle 2 Pro and Enter 3 parts

### Electronics

BTT SKR MINI E3 V3.0

### LCD/Controller

BTT MINI 12864 V2.0

### Other add-ons

SKR MINI SCREEN ADAPTER

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Configuration and pin dump attached. 
[config.zip](https://github.com/MarlinFirmware/Marlin/files/15368797/config.zip)

The board changes for the LCD SD card are:  
//The following block selects the onboard SD card.  Comment it out.
/*
#if SD_CONNECTION_IS(ONBOARD)
  #define SD_DETECT_PIN                     PC3
#elif SD_CONNECTION_IS(LCD) && (ALL(TOUCH_UI_FTDI_EVE, LCD_FYSETC_TFT81050) || IS_TFTGLCD_PANEL)
  #define SD_DETECT_PIN              EXP1_01_PIN
  #define SD_SS_PIN                  EXP1_05_PIN
#elif SD_CONNECTION_IS(CUSTOM_CABLE)
  #error "SD CUSTOM_CABLE is not compatible with SKR Mini E3."
#endif

#define ONBOARD_SPI_DEVICE                     1  // SPI1 -> used only by HAL/STM32F1...
#define ONBOARD_SD_CS_PIN                   PA4   // Chip select for "System" SD card
*/
//This points at the LCD SD cards new pins
#define SD_DETECT_PIN              PD5
#define ONBOARD_SD_CS_PIN          PD4

//The following line was added to the SPI setup for STM32
OUT_WRITE(EXP1_05_PIN, HIGH);

---

Pinout through the adapter and showing the two new connections:
[image]

RzTen1, 2024-05-19:
That line at the bottom should read OUT_WRITE(PA4, HIGH) instead of OUT_WRITE(EXP1_05_PIN, HIGH);  I accidently copied that from an earlier test.

thisiskeithb, 2024-05-19:
I also ran into this with the same LCD, motherboard, and custom wire harness (and eventually LCD adaptor board from Timmit99).

I ended up swapping to a TFT35, but the issue remained for this hardware combination.

RzTen1, 2024-05-20:
I think I've figured it out.  It looks like the transition from SPI mode 0 to mode 3 is happening too fast for the screen to pick up, even in software SPI mode.  I've made some tweaks to the way SPI works with this display and adapter in the STM32 source and everything appears to be working correctly.  I can browse the SD card and start a print without the screen locking up.  It still gets stuck on the Marlin boot screen if a card is inserted when the printer powers up, but clicking the button brings up the menu like normal so I'm not sure if this is a bug with the screen or a more general problem. In either case I can live with it. I'll try to generate a pull request with the changes.

RzTen1, 2024-05-21:
Actually this may be a bit premature.  It looks like the screen is still stalling out on some SD card reads.  I'm able to clear the screen by hitting the button, but I have to do it after every card read when it's printing.  Let me dig into it a bit more.  It's still better than it was but I wouldn't call it fixed.

RzTen1, 2024-05-24:
The PR should be all set to go.  It looks like there still may be an issue with other STM32G0 boards but it at least fixes the SKR Mini E3 V3.

closed 2024-06-08 by thinkyhead