MarlinFirmware/Marlin #27104: [BUG] After completing Home of all three axis, the Z axis rises up about 10mm

dmarnel, 2024-05-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Not sure when this started happening, but after pressing Home on my Ender 3 Pro, and after all three axis have finished homing, the Z position of the print head rises up about 10mm and stops there.

If I uncomment Z_AFTER_HOMING in Configuration.h, and set it to zero, the Z axis no longer rises up after homing is completed.

### Bug Timeline

Unsure

### Expected behavior

I expect all three axis to remain in the "home" position after selecting the Home option from the printer display.

### Actual behavior

The Z axis moves up about 10mm and remains there.

### Steps to Reproduce

1. Press the Home button
2. Wait for all three axis to finish homing
3. Z axis moves up about 10mm

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Creality Ender 3 Pro

### Electronics

BTT SKR MINI E3 V2.0

### LCD/Controller

BTT TFT35-E3 V3.0

### Other add-ons

Raspberry Pi connected through USB port for WiFi & Octoprint

### Bed Leveling

None

### Your Slicer

Simplify3D

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/15369836/configs.zip)

ellensp, 2024-05-19:
Please read the description for Z_AFTER_HOMING   "(mm) Height to move to after homing (if Z was homed)"

ie this is doing exactly what it is meant to do.

ellensp, 2024-05-19:
I read this backwards

this is happening with Z_AFTER_HOMING disabled

thisiskeithb, 2024-05-19:
`Z_AFTER_HOMING` is set as follows:

https://github.com/MarlinFirmware/Marlin/blob/06762db050399137796bbd192b94ed9da4a77e11/Marlin/src/module/motion.h#L417-L421

Since `Z_AFTER_HOMING` is not defined, it'll set `Z_POST_CLEARANCE` to `Z_CLEARANCE_FOR_HOMING`.

You have disabled `Z_CLEARANCE_FOR_HOMING` and `Z_CLEARANCE_BETWEEN_PROBES` is set to `5` in your config, so `Z_CLEARANCE_FOR_HOMING` it set to `5` (`Z_CLEARANCE_BETWEEN_PROBES`):

https://github.com/MarlinFirmware/Marlin/blob/06762db050399137796bbd192b94ed9da4a77e11/Marlin/src/inc/Conditionals_post.h#L3215-L3221

Z should raise by 5, not 10.

closed 2024-05-19 by thisiskeithb

dmarnel, 2024-05-19:
Which is exactly what I said I had to do to keep it from raising the Z axis. BTW, I haven't disabled/commented out anything having to do with homing, that is how it was set when I pulled the new changes It's been about a year since I've messed with the code. Regardless, it doesn't make sense to have to uncomment/change something to keep the axis in the home position. Maybe Z_AFTER_HOMING should be uncommented and set to zero as a default setting?