MarlinFirmware/Marlin #27126: [BUG] Erratic extruder action

rq3, 2024-05-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The latest bugfix (May 24, 2024) now works on my delta. Calibration and M48 now works, BUT the extruder is very erratic.

### Bug Timeline

New

### Expected behavior

The extuder should respond to the g-code commands.

### Actual behavior

While the x, y, and z axes seem to be functional, the extruder acts as if it is receiving enable pulses that are too short, or as if its current is too low. The extruder is driven by an A4988, so this is not a driver issue. The extruder randomly vibrates, moves forward, then backwards (even though there are no retracts called for).

Earlier Marlin works fine with the same g-code.. This is a recent bugfix problem.

### Steps to Reproduce

1) Compile and install latest Marlin
2) Print previously good g-code
3) See extruder not respond properly

### Version of Marlin Firmware

latest bugfix (May 24, 2024)

### Printer model

Anycubic Predator

### Electronics

stock Trigorilla Pro STM32f103ZETA

### LCD/Controller

Anycubic 3.5 TFT

### Other add-ons

none

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

`
[Configs.zip](https://github.com/MarlinFirmware/Marlin/files/15440252/Configs.zip)

https://github.com/MarlinFirmware/Marlin/assets/3343927/682405bf-6ee2-401b-aabb-68f29c650d23

`

classicrocker883, 2024-05-25:
that jump scare at the end...
could this be a hardware issue? such as the wires or pin connectors?

have you changed anything regarding the motor recently?

does a previous firmware version work?

rq3, 2024-05-25:
> that jump scare at the end... could this be a hardware issue? such as the wires or pin connectors?
> 
> have you changed anything regarding the motor recently?
> 
> does a previous firmware version work?

I regressed to a bugfix from January 12, which works fine. Definitely a Marlin issue.

ellensp, 2024-05-27:
well you have a commit that works and one that doesn't, I suggest you try a git bisect and track down the exact commit that changed the normal behavior.

rq3, 2024-05-27:
> well you have a commit that works and one that doesn't, I suggest you try a git bisect and track down the exact commit that changed the behavior.

Not practical. There is a roughly 4 month window between January and May of this year when the Marlin bugfixes did not work at all on my delta. Only with last weeks bugfix did the machine actually begin to try to operate properly, but still has extrusion issues.

So I can say that the January 12 bugfix worked, and everything after that was unuseable until last week.

classicrocker883, 2024-05-27:
well if the Jan 12 bugfix worked then you can try to either check the next commit which may be the issue in question, compile flash and test from there. it would be tedious to do this for every other commit, BUT what you *can* do (as I've done) was pick a bugfix halfway between it started acting this way, to the one that actually worked well.

if you do this and split each timeframe in half you may only need to do this several times. unless you get lucky.

ellensp, 2024-05-28:
Git bisect is well designed, it eliminates half the commits every iteration. you would only have to test approximately 7 times ...

Data we have
good [cron] Bump distribution date (2024-01-12) 46f370ae3cc1501bd831c495b46ab0e571e99f10
bad [cron] Bump distribution date (2024-05-27) e7c9cf3e1db42f90ef2b355b86faefe91b809427

Ie to do a git bisect

```
git bisect start
git bisect good 46f370ae3cc1501bd831c495b46ab0e571e99f10
git bisect bad e7c9cf3e1db42f90ef2b355b86faefe91b809427
```
Which tells you "Bisecting: 149 revisions left to test after this (roughly 7 steps)"

There are only 300 commits and 78 of those are just cron updating the distribution date

Ie it has to be in one of these

<details>
  <summary>Click me</summary>

commit 6710616a090d66d73e4dc6b09f38f86ae63c50f5
Author: ellensp <530024+ellensp@users.noreply.github.com>
Date:   Mon May 27 05:09:18 2024 +1200

    🩹 Longer3D LK has ONBOARD_SDIO (#27129)

commit 2064c83c6681afac7dec636ce361e91b8ecba76b
Author: Scott Lahteine <thinkyhead@users.noreply.github.com>
Date:   Sun May 26 11:57:42 2024 -0500

    🚸 Fix SD nav after "one click print" (2)

commit 4f85f88ae3a8424b17b42c59f53e09647c046c0c
Author: Scott Lahteine <thinkyhead@users.noreply.github.com>
Date:   Sun May 26 02:12:22 2024 -0500

    🚸 Fix SD nav after "one click print"

commit 3ee1248cf2183ab8e733fb5a92d1e85728744ecd
Author: Keith Bennett <13375512+thisiskeithb@users.noreply.github.com>
Date:   Wed May 22 19:58:22 2024 -0700

    🩹 Even more Z input shaper followup (#27119)

    Followup to #27118

commit e3bbacd8b928a73004494b216aaf8655b2134421
Author: Keith Bennett <13375512+thisiskeithb@users.noreply.github.com>
Date:   Wed May 22 16:30:44 2024 -0700

    📝 Update Input Shaping comments (#27116)

    Co-authored-by: Scott Lahteine <thinkyhead@users.noreply.github.com>

commit f9153712483bf6ad3322de3ab4adf5c55bbe47a9
Author: Scott Lahteine <thinkyhead@users.noreply.github.com>
Date:   Wed May 22 17:36:44 2024 -0500

    🧑‍💻 More language cleanup

commit ea848aa891808c27d0fc0039a93bb454cf81daf6
Author: Andrew <18502096+classicrocker883@users.noreply.github.com>
Date:   Wed May 22 15:07:04 2024 -0400

    🩹 Z input shaper followup (#27118)

    Followup to #27073

[... 1488 lines cut ...]

commit 3019af1c9398da0172b5d5f5bb0d3a5506e56e4d
Author: Scott Lahteine <thinkyhead@users.noreply.github.com>
Date:   Sat Jan 13 18:30:43 2024 -0600

    🔨 Make / pins-format patches

commit b2fd631d82b9b3e64226ceac6a132ec917aaee21
Author: Robherc <68039049+robherc@users.noreply.github.com>
Date:   Fri Jan 12 23:03:34 2024 -0500

    🔧 Fix SD connection defaults (#26666)

    Co-authored-by: thisiskeithb <13375512+thisiskeithb@users.noreply.github.com>
    Co-authored-by: Scott Lahteine <thinkyhead@users.noreply.github.com>

commit ab3497161a37ae309e030a23e0f8e3f3eb61daea
Author: ThomasToka <117008525+ThomasToka@users.noreply.github.com>
Date:   Fri Jan 12 06:56:45 2024 +0100

    🐛 Fix PLR pos/sdpos (#26365)

    Co-authored-by: Scott Lahteine <thinkyhead@users.noreply.github.com>

</details>

ellensp, 2024-05-28:
Can you provide a test gcode that exhibits the issue?

rq3, 2024-05-28:
> Can you provide a test gcode that is known to exhibit the issue?

Sliced in Cura 5.5.0 and used in both of the above videos.
[Test_Cube.zip](https://github.com/MarlinFirmware/Marlin/files/15470681/Test_Cube.zip)

rq3, 2024-05-28:
> Git bisect is well designed, it eliminates half the commits every iteration. you would only have to test approximately 7 times ...
> 
> Data we have good [cron] Bump distribution date (2024-01-12) [46f370a](https://github.com/MarlinFirmware/Marlin/commit/46f370ae3cc1501bd831c495b46ab0e571e99f10) bad [cron] Bump distribution date (2024-05-27) [e7c9cf3](https://github.com/MarlinFirmware/Marlin/commit/e7c9cf3e1db42f90ef2b355b86faefe91b809427)
> 
> Ie to do a git bisect
> 
> ```
> git bisect start
> git bisect good 46f370ae3cc1501bd831c495b46ab0e571e99f10
> git bisect bad e7c9cf3e1db42f90ef2b355b86faefe91b809427
> ```
> 
> Which tells you "Bisecting: 149 revisions left to test after this (roughly 7 steps)"
> 
> There are only 300 commits and 78 of those are just cron updating the distribution date
> 
> Ie it has to be in one of these
> Click me

I'm all too familiar with bisecting releases, but the release has to actually compile and run (at least to some extent) on the printer. Until just last week, that was not the case for any bugfix after Jan. 12 for my delta printer.

classicrocker883, 2024-05-29:
I suppose then there may be more than one issue. 
1. Whatever is causing this extruder issue
2. Whatever is giving you the compile error

can you post what sort of errors youre getting that isnt letting you compile?

rq3, 2024-06-01:
The 2.1.x bugfix of May 30 still shows this erratic extrusion. However, the patched 2.1.2.3 release not only works fine but also has all three digits after the decimal in the standard deviation displays (a seperate but concerning issue).

thisiskeithb, 2024-08-30:
> The 2.1.x bugfix of May 30 still shows this erratic extrusion.

Is this still an issue?

closed 2024-09-08 by rq3