MarlinFirmware/Marlin #27139: [FR] (feature summary) TOOLHEAD E3D changing for stepper instead servo

Eyolon, 2024-05-29:
### Is your feature request related to a problem? Please describe.

Hi

I try to code the feature by myself and it's highly based on the SWITCHING_TOOLHEAD section in the tool change module. (basicaly samen just need to change the MOVE_SERVO call by a rotation of an extruder motor, 0 to 90 deg, with the gear ratio, gear diameter etc...

My only problem is after many hours to read the code i havent find a simple function called like "move_stepper" or others "rotate_extruder"

So i need help for the motion part, i've try to set destination.e and then call prepare_line_to_destination() but it doesn't works and i dont understand the logic behind code
i just want to "move the stepper E0 to X steps" with kiss code

### Are you looking for hardware support?

RAMPS 1.4, Ard2560, Marlin 2.0.7.3

### Describe the feature you want

As you can read here https://e3d-online.com/blogs/news/toolchanger-the-update-youve-all-been-waiting-for and see on many demo video, the E3D tool changer does not use servo for the locking system anymore

DerAndere1, 2024-06-01:
You can define the stepper that drives the locking-mechanism of the toolcanger as an extra axis (enable and set `I_DRIVER_TYPE` and `#define AXIS4_NAME 'A'`in configuration.h etc.). Then you can use `EVENT_GCODE_TOOLCHANGE_T0`, `EVENT_GCODE_TOOLCHANGE_T1`, etc in Configuration_adv.h instead of modifying toolchange.cpp to define the toolchange sequence. If you cannot use EVENT_GCODE_TOOLCHANGE_T0, do not change the `destination` variable. `destination`is the position where the printer willl go to AFTER the toolchange is completed. So you shou instead set the variable `current_position` (for the extra axis: `current_pos.i = ...`. Then use one of the variants of the function `line_to_current_position()` (e.g. `fast_line_to_current()` or `slow_line_to_current()`. Also, test whether or not you need to add the function `planner.synchronize()`

Eyolon, 2024-06-08:
Hi

Thank's for your answer. i've faced to electrical problem like the motor is not 12V but i guess 9V (come from old paper printer so i try to not burn it x) ) but now i can try your proposition, i quickly answer if it works for me

Eyolon, 2024-06-13:
Hi there
after update, try your solution and face to logical problem like "i enable I axis but i cant use toolhead switch because i havent 2 extruder and now i've set 2 extruder, i havent motors anymore for I... and i just want one extruder with my laser...."

So i finaly make it works and i've push on a repository on my github account to keep difference between 2.1.1 lts vanilla and with my change to use a E3D like with stepper motor, ONE extruder and a laser (and maybe spindle in future)

check it here https://github.com/Eyolon/Marlin-Cube/commit/031991333e634b7ef1e04e7f36e61491020074dc if you see something usefull and thank's again, AND for your work on Marlin, AND for your precious answer and attention

DerAndere1, 2024-06-14:
It is good to know that only those few changes were necessary to support a mix of LASER_FEATURE/SPINDLE_FEATURE and an extruder. If you have enabled HOTEND_OFFSETS, test that feature carefully. You might need some changes from my open pull request https://github.com/MarlinFirmware/Marlin/pull/26956 to make HOTEN_OFFSETS work with your setup