MarlinFirmware/Marlin #27140: Compile error SDSUPPORT Artillery Ruby

pmanzionna, 2024-05-30:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

[image]

### Bug Timeline

New bug with Marlin 2.1.2.3

### Expected behavior

Compile error

### Version of Marlin Firmware

2.1.2.3

### Printer model

Artillry Hornet

### Electronics

Artillery_ruby

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

pmanzionna, 2024-05-30:
official config/examples/Artillery/Hornet files

pmanzionna, 2024-05-30:
Thank you ellensp... it works!

ellensp, 2024-05-30:
I think it was ment to to be #elif HAS_MEDIA  but that needs testing

pmanzionna, 2024-05-30:
Tested with #elif HAS_MEDIA... it seems to work.