MarlinFirmware/Marlin #27141: [BUG] DWIN_LCD_PROUI won't draw screen

lonelyhero77, 2024-06-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

DWIN_LCD_PROUI is not working

### Expected behavior

The DWIN LCD should be working

### Actual behavior

The DWIN LCD won't draw anything except showing black screen

### Steps to Reproduce

1. define DWIN_LCD_PROUI in /Marlin/Configuration.h
2. Build the firmware
3. Upload to the machine
4. Update the machine

### Version of Marlin Firmware

head of bugfix-2.1.x branch

### Printer model

Ender-3 V2

### Electronics

BOARD_CREALITY_V422

### LCD/Controller

Stock Ender-3 V2 DWIN LCD

### Other add-ons

BLTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I use OrcaSlicer(forked from PrusaSlicer) for slicing
[configuration.zip](https://github.com/user-attachments/files/15522700/configuration.zip)

ellensp, 2024-06-02:
Did you actually update the firmware in the display?  (this is separate from the controller firmware)
https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Creality/Ender-3%20V2

thisiskeithb, 2024-06-02:
I'm going to mark this as confirmed since someone else on Discord is having the same issue with ProUI, even with the updated/matched DWIN set.

MarlinUI works fine.

lonelyhero77, 2024-06-02:
> Did you actually update the firmware in the display? (this is separate from the controller firmware) https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Creality/Ender-3%20V2

Yes, my DWIN Display is up to date.

classicrocker883, 2024-06-02:
I address this issue in this https://github.com/MarlinFirmware/Marlin/pull/26917

- ProUI starting up has issues, reverted SETUP_RUN(...) in MarlinCore.cpp for fix

**MarlinCore.cpp**
```diff
#if HAS_DWIN_E3V2
  #include "lcd/e3v2/common/encoder.h"
  #if ENABLED(DWIN_CREALITY_LCD)
    #include "lcd/e3v2/creality/dwin.h"
  #elif ENABLED(DWIN_CREALITY_LCD_JYERSUI)
    #include "lcd/e3v2/jyersui/dwin.h"
+ #elif ENABLED(DWIN_LCD_PROUI)
+   #include "lcd/extui/ui_api.h"
  #endif
#endif

	@@ -1599,6 +1601,8 @@ void setup() {

  #if ENABLED(DWIN_CREALITY_LCD)
    SETUP_RUN(dwinInitScreen());
+ #elif ENABLED(DWIN_LCD_PROUI)
+   SETUP_RUN(ExtUI::onStartup());
  #endif

lonelyhero77, 2024-06-05:
> I address this issue in this #26917
>
> * ProUI starting up has issues, reverted SETUP_RUN(...) in MarlinCore.cpp for fix
>
> **MarlinCore.cpp**
>
> ```diff
> #if HAS_DWIN_E3V2
>   #include [...]

I've merged your PR to my up to dated fork. It was automerged but some errors ar occurred during build.

line 1235 @ /Marlin/src/lcd/e3v2/proui/dwin.cpp 
```
switch (hmiValue.tempControl) {
          #if ENABLED(PIDTEMP)
            case PIDTEMP_START: drawHPlot(); break;
          #elif ENABLED(MPCTEMP)
            case MPCTEMP_START: drawHPlot(); break;
          #endif
          #if ENABLED(PIDTEMPBED)
            case PIDTEMPBED_START: drawBPlot(); break;
          #endif
          #if ENABLED(PIDTEMPCHAMBER)
            case PIDTEMPCHAMBER_START: drawCPlot(); break;
          #endif
        default: break;
        } break;
```
The scope MPCTEMP_START should be changed to MPC_STARTED. (this one is the only one which hasn't changed)

After I've changed the name of scope, The compiler shows this error: `Marlin\src\lcd\e3v2\proui\proui_extui.cpp:220:45: error: cannot convert 'ExtUI::mpcresult_t' to 'tempcontrol_t'. I've no idea of this error maybe there's something that I mistake.
`

lonelyhero77, 2024-06-06:
@classicrocker883 I've found that the PR #26917 using the scopes that are not declared in current upstream which occurs error during build. Such as dwinMPCTuning() to DWIN_MPCTuning() so on.... I believe the PR should also include the commit which includes changing name of scopes or should discuss which naming convention is okay to use for this project.

classicrocker883, 2024-06-08:
youre right thanks

nicwilson58, 2024-06-16:
when is it estimated the fix for this will appear in the nightly build?

thisiskeithb, 2024-06-16:
> when is it estimated the fix for this will appear in the nightly build?

There is no estimated time at the moment. I'd recommend using `DWIN_MARLINUI_PORTRAIT` or one of the other DWIN UIs if you want to run current `bugfix-2.1.x` code since `DWIN_LCD_PROUI` is still causing lockups/no boot scenarios.

classicrocker883, 2024-06-18:
@lonelyhero77 did that fix work for you, so everything works as it should?

lonelyhero77, 2024-07-06:
> @lonelyhero77 did that fix work for you, so everything works as it should?

Sorry. I haven't been able to test in real for a while. I tested it now and it seems working OK. It now draws screen. but the firmware is unable to load dwin icons. Maybe it's related to the code that are about allocating dwin ico number.... I'm not sure yet.

classicrocker883, 2024-07-07:
you probably have a different .ico file installed than defined.
check your **DWIN_SET** which you installed in the LCD, you should see 9.ICO or 7.ICO (could be something else)

in **Marlin\src\lcd\e3v2\common\dwin_set.h** look for `#define ICON`

change this to:  
`#define ICON 9` if you have 9.ICO in the **DWIN_SET**

lonelyhero77, 2024-07-19:
> you probably have a different .ico file installed than defined. check your **DWIN_SET** which you installed in the LCD, you should see 9.ICO or 7.ICO (could be something else)
>
> in [...]

This worked for me. Thanks for the contribution!

I hope the PR gets merged soon.

I've tested it on MKS Robin Nano V3.1 and Creality V4.2.2 Board.

lonelyhero77, 2024-07-20:
@classicrocker883 I've found that the screen still won't draw when I turn on AUTO_BED_LEVELING_UBL function.  I've no idea what's related to the screen drawing but if I turn off UBL function the screen works great. Do you have any clue about this happen?

classicrocker883, 2024-08-02:
IIRC, I think I experienced the same issue. It can be related to **settings.cpp**, I mean that is my hunch.

the main issue basically stems when ProUI was lumped into `EXTENSIBLE_UI`. it's also possible for something in **proui_extui.cpp**

but when the screen won't show up because of the bed leveling it's probably something to do with settings.cpp

nicwilson58, 2025-01-12:
I see this issue is still open.  is it actively being worked on,  I have some users requesting firmware with ProUI

classicrocker883, 2025-01-12:
then I've got some good news for you!
well, here are some compilations I've been working on ever since, so if you have this type of printer this will work fine - [MRiscoCProUI](https://github.com/classicrocker883/MRiscoCProUI/releases).

otherwise if you want a more... vanilla plain version I've been working on a pull request that so far fixes the issues, [you can find it here](https://github.com/MarlinFirmware/Marlin/pull/26917)

or just go to the [repo address](https://github.com/classicrocker883/Marlin/tree/bugfix-2.1.x-March2), copy the zip file.

closed 2026-10-23 by thinkyhead