MarlinFirmware/Marlin #27151: [Sensorlees homing] (TMC2209 homing brake after BACKOFF)

telnet01, 2024-06-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have a five motors 17HS4401 with TMC2209 on scr 1.4. All motors operate correctly when performing linear motion (G0 command). The first four motors have a speed reduction of more than 4 and homing is correctly, unlike the last one. The last  one  (J motor) has a speed reduction of 1 to 1 and stops when the "G28 B" command is called immediately after BACKOFF. I tried to change microstepings,travel speed, homing speed, backoff bump, change current. This didn't work. Helps only if I change speed reduction more then twice for example (DEFAULT_AXIS_STEPS_PER_UNIT parameter, currect is 8.88 (200*16/360) for my configuration). but it doesn't suit my mechanics. If i change J_STALL_SENSITIVITY to 1..5 it start homing, but then when encounters an obstacle, he does not stop but tries to break through it, so that the shaft begins to spin.

 ```
 #if AXIS_IS_TMC_CONFIG(J)
    #define J_CURRENT      600
    #define J_CURRENT_HOME J_CURRENT
    #define J_MICROSTEPS    16
    #define J_RSENSE         0.11
    #define J_CHAIN_POS     -1
    //#define J_INTERPOLATE  true
    //#define J_HOLD_MULTIPLIER 0.5
  #endif
#define SENSORLESS_HOMING // StallGuard capable drivers only

  #if EITHER(SENSORLESS_HOMING, SENSORLESS_PROBING)
    // TMC2209: 0...255. TMC2130: -64...63
    #define X_STALL_SENSITIVITY  80
  //  #define X2_STALL_SENSITIVITY X_STALL_SENSITIVITY
    #define Y_STALL_SENSITIVITY  60
  //  #define Y2_STALL_SENSITIVITY Y_STALL_SENSITIVITY
    #define Z_STALL_SENSITIVITY  80
    //#define Z2_STALL_SENSITIVITY Z_STALL_SENSITIVITY
    //#define Z3_STALL_SENSITIVITY Z_STALL_SENSITIVITY
    //#define Z4_STALL_SENSITIVITY Z_STALL_SENSITIVITY
    #define I_STALL_SENSITIVITY  40
    #define J_STALL_SENSITIVITY 40
    //#define K_STALL_SENSITIVITY  8
    //#define U_STALL_SENSITIVITY  8
    //#define V_STALL_SENSITIVITY  8
    //#define W_STALL_SENSITIVITY  8
    //#define SPI_ENDSTOPS              // TMC2130/TMC5160 only
    //#define IMPROVE_HOMING_RELIABILITY
  #endif

#define SENSORLESS_BACKOFF_MM  { 5, 5, 5 ,5,10 }  // (linear=mm, rotational=°) Backoff from endstops before sensorless homing
#define HOMING_BUMP_MM      { 10, 0, 0 , 0, 0 }       // (linear=mm, rotational=°) Backoff from endstops after first bump
#define HOMING_BUMP_DIVISOR { 10, 10, 4 ,10 ,2 }       // Re-Bump Speed Divisor (Divides the Homing Feedrate)

#define DEFAULT_AXIS_STEPS_PER_UNIT   { 44.44 , 35.55, 100, 9.99, 8.88 }
#define HOMING_FEEDRATE_MM_M { (20*60), (30*60), (10*60) ,(60*60), (40*60)}

#define J_MIN_POS 0
#define J_MAX_POS 360
```

```
M122
		X	Y	Z	I	J
Address		0	0	0	0	0
Enabled		false	false	false	false	true
Set current	850	850	850	850	600
RMS current	826	826	826	826	581
MAX current	1165	1165	1165	1165	819
Run current	26/31	26/31	26/31	26/31	18/31
Hold current	13/31	13/31	13/31	13/31	9/31
CS actual	13/31	13/31	13/31	13/31	18/31
PWM scale					
vsense		1=.18	1=.18	1=.18	1=.18	1=.18
stealthChop	true	true	true	true	true
msteps		4	4	4	4	16
interp		false	false	false	false	false
tstep		max	max	max	max	max
PWM thresh.					
[mm/s]						
OT prewarn	false	false	false	false	false
triggered
 OTP		false	false	false	false	false
pwm scale sum	15	15	15	15	25
pwm scale auto	0	0	0	0	0
pwm offset auto	36	36	36	36	45
pwm grad auto	14	14	14	14	14
off time	4	4	4	4	4
blank time	24	24	24	24	24
hysteresis
 -end		2	2	2	2	2
 -start		1	1	1	1	1
Stallguard thrs	80	60	80	40	40
uStep count	32	32	32	32	8
DRVSTATUS	X	Y	Z	I	J
sg_result	0	0	0	0	16
stst						
olb						
ola						
s2gb						
s2ga						
otpw						
ot						
157C						
150C						
143C						
120C						
s2vsa						
s2vsb						
Driver registers:
		X	0xC0:0D:00:00
		Y	0xC0:0D:00:00
		Z	0xC0:0D:00:00
		I	0xC0:0D:00:00
		J	0xC0:0C:00:00

Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing I connection... OK
Testing J connection... OK
ok

```

### Version of Marlin Firmware

Marlin 2.1.2.2 

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/user-attachments/files/15551720/config.zip)

ellensp, 2024-06-06:
Configs are for 2.1.2.2

Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

telnet01, 2024-06-07:
> Configs are for 2.1.2.2
> 
> Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Hello, I deployed this assembly in a minimal configuration for testing - I connected the motor to the X-axis. But I could not conduct research, since after homing it immediately occurs system halted. This happens even when sensorles_homing is disabled and is triggered immediately after the software limit on the X-axis values ​​is exceeded. If sensorles_homing is enabled - behavior as in the topic of the post, and then halted immediately after stopping the motor.
```
>>> G28 X
SENT: G28 X
RECV: echo:busy: processing
RECV: echo:Homing Failed
RECV: Error:Printer halted. kill() called!
Error:Printer halted. kill() called!
```

closed 2024-10-11 by github-actions[bot]