MarlinFirmware/Marlin #27156: [BUG] layer shift when switching extruders and slow layer shift during long prints

Legion23705, 2024-06-09:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I discovered 2 malfunctions regarding the positioning of one axis. The error has been detected on the X-Axis, but might also occur on the other axes (not testet jet) 

1: Axis shifts after tool change
When I perform a tool change, the real position of the axis shifts a little bit. 
For context: My printer has 2 tool heads which are seperated along Y by 61,64mm and along X by 0mm. These values are compensated by the slicer. However, there is always an additional offset which changes e.g. when I change the nozzle, (currently 0,55 for X, 0,1 for Y). This additional offset is stored in the Marlin EEPROM, so that the axes move by this amount when I switch the toolhead. This works generally. 
I use closed loop servo motors for the X, Y and Z-Axes (Leadshine iSV2), so step losses by the motor are not possible. The extruder motors use external high power stepper drivers from leadshine.
The positions have been measured both with the motor parameters (command step value) and with a calliper on the printer, so it is not a mechanical problem (loose couplings, belts or whatever).

The problem is that after I switch from T0 to T1 and back to T0, the axes don´t stay at their original position. I only tested this for the X axis, but it may be present at the Y axis as well. 
What makes this phenomenom even stranger is that this position shift depends on the actual axis position.
I wrote a small test programm to perform 100 toolchanges (
[Test Tool Change x100.zip](https://github.com/user-attachments/files/15751401/Test.Tool.Change.x100.zip) )
 rapidly to make the shift more visible, here are the results: 

at X0: shift X +0,86mm
at X1: -0.85mm
at X5: -0.85mm
at X10: -0,85mm
at X20: -0.85mm
at X30: +0.85mm
at X40: +0.85mm
at X50: +-0mm
at X100: +0,0025mm
at X400: +0,0025mm
at X800: +0,0025mm

These results are reproducible and consistent, so at X0 each tool change shifts the axis by 0,0085mm (which is exactly 1 step).
Since the layer shift depends on the position, it can only be a firmware issue. 
I also tested if the shift happens when I only move the axis by 0,55mm back and forth 100 times without tool change: In this case the position does not change. So it seems to only happen when the tool changes. 
Maybe the step generation during the tool change is broken (too short signal or too small delay between direction change and step), but I can´t test this because I only have a cheap oscilloscope and no logic analyser.

2: Slow axis shift during long prints
I have noticed that the X-axis slowly shifts as well during long prints when I dont´t switch the toolhead (I noticed it at an 48h-printjob). Maybe both problems have the some cause, so I wrote them together in this report. 
The axis shifts by about 0,5mm every 24 h of print time +-. 
This is about 1 - 2 Steps every hour. In result the axis always shifts in X+, but I also measured small shifts in X- by a few steps. 

Here is my config: 
[Configuration.zip](https://github.com/user-attachments/files/15751354/Configuration.zip)
I didn´t test the version 2.1.2.3 yet because I have to keep the machine running for an order as long as possible and currently have only limited time for testing. I checked the latest release notes and bug reports and since there is no mention of such a problem, I don´t think that it has been found or fixed since 2.1.2.1.

I had the theory that at least the problem no. 2 could be caused by emc problems. My printer is quite large and the cables are long. In the past I had EMC-problems at the Y axis which has the longes cables (which was clearly visible on the oscilloscope) which also resulted in a position shift. I fixed this problem by using better shielded cables and seperating power and signal cables as much as possible. When this problem was acute, I could see the axis jump by a few steps every time I power up the extruder stepper motors, but this is not the case this time. 

The step signal looks quite clear, the noise is below 1V, so I believe it is not an EMC problem this time. 
[image]

### Bug Timeline

I first noticed it a few weeks ago

### Expected behavior

Consistent positioning after tool change

### Actual behavior

Position shift after tool change and during long prints

### Steps to Reproduce

Switch toolheads when a toolhead offset is set in EEPROM

### Version of Marlin Firmware

2.1.2.1

### Printer model

custom built

### Electronics

BIGTREETECH Manta M8P with Raspberry Pi 4

### LCD/Controller

-

### Other add-ons

closed loop servo motors and external steper drivers

### Bed Leveling

No Bed Leveling

### Your Slicer

Simplify3D

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Host Software: Repetier Server

thisiskeithb, 2024-06-09:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

Legion23705, 2024-06-12:
I´ve tried the bugfix-2.1.x firmware:
[Configuration.zip](https://github.com/user-attachments/files/15803889/Configuration.zip)

The problems are both still present, but changed a little.
The position shift still happens after a tool change, but the values for direction and starting position have changed:

Position shift comparison after 100 toolchanges:
starting position | Firmware   2.1.3.x bugfix | Firmware   2.1.2.1

X0  | -0.85  | 0,85
X1  | -0.85  | -0,85
X2  | -0.85  | n.a.
X3  | +0.85 | n.a.
X4  | 0         | n.a.
X5  | 0         | -0,85
X6  | -0.85   | n.a.
X7  | +0.85  | n.a.
X8  | +0.85  | n.a.
X9  | 0         | n.a.
X10 | 0       | -0,85
X20 | -0.85 | n.a.

I also discovered that the Y-axis doesn´t shift after toolchange at any position, only the X-axis is affected. 
The physical position of the axes seemingly don´t matter: When I home the X-axis at a different location by manually hitting the endstop and do the tests again, the shifting pattern is still the same. 

The problem with the consistent slow layer shift during a large print hasn´t changed with the new firmware. 
This problem occurs also at the Y-axis.

I tried to test if somethings wrong with the timing of Step + direction. 
This scope shows a fast back - and forth movement by 0,5mm, yellow is step, blue is direction (the voltage is inverted):

Scale 5 µs
[image]

Apparently the timing looks fine, the step width is 4,5µs (minimum: 2,5µs according to leadshine) and the distance between direction change and step is 15 µs (minimum 2,5 according to leadshine). 

Scale 500 µs
[image]

It is also visible that the signal takes some time to return to 0V and doesn´t go below 1V at higher frequencies. 
The leadshine servos use optocouplers to read the signal and I think that there is almost no current through the LEDs at 1V which leads to the slow voltage drop, I assume that this 1V voltage level is read as zero anyway and therefore shouldn´t cause the problem.

I did the same for the tool change: 
[image]

The time distance between step and direction signals are even larger here, so there is no obvious problem with the signals.

Legion23705, 2024-08-11:
Are there any updates to this problem? 
It wasn´t solved yet as far as I know.

thinkyhead, 2024-08-11:
Have you tried adjusting the settings for stepper signal timings to see if they have any effect on the amount of error? We've been adjusting the stepper base timings according to our best estimates and measurements, but they may still be on the edge. So, slightly lengthening the duration of STEP signals and the delay after DIR signals may have some noticeable benefit.

Legion23705, 2024-08-14:
I repeated the same test (100 x T0 T1) with increased MINIMUM_STEPPER_PULSE_NS 3000 ->10000, MINIMUM_STEPPER_PRE_DIR_DELAY + POST_DIR_DELAY 5000 -> 10000: Same result at position X1 (Shift X-0,85)
I also changed the acceleration value (Acc XY 4000 -> 1000, Jerk XY 10 -> 5): Same result at position X1 (Shift X-0,85). 

However, at position X4 for example everything is fine. The X-Axis doesn´t shift when I repeat the test here. 
So as a workaround, I can just edit the tool change macro so that it executes the toolchange at X4. 

The other problem (consistent slow layer shift during a large print, without tool change) has been solved: It turned out that the filament inside the PTFE-Tube, which runs parallel to the signal and power cables, was building up electrostatic charges which discharged from time to time, sometimes even with a visible flash. This discharge created a small electromagnetic pulse which was picked up by an unshielded power cable. This pulse must have created a small voltage spike which was misinterpreted as a step command. 
I discovered this because the layer shift went worse when I printed with a higher filament throughput, which created more pressure at the inside of the PTFE tube and so increased the static charge buildup. 
I replaced the PTFE-Tube with an antistatic PTFE-Tube and replaced the unshielded with a shielded cable. 
This solved this problem, but the toolchange problem is still present.

closed 2024-11-27 by github-actions[bot]