MarlinFirmware/Marlin #27159: MKS H43 + Creality 4.2.7 | MKS H43 doen't work proberly

mlueft, 2024-06-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am using a MKS H43 with the creality 4.2.7 board.
Everything is working fine. I get the status screen and it shows the bed and extruder temperature.
But I can't do anything on the H43, changing screens.
As far as I understood the communication is working, because I can enter the language menu.
But I can't change to english or do anything else like homing er moving the print head.

I studied all Configuration values but nothing works.

Has anyone this combination working, or anyone an idea what I am doing wrong?

MKS H43 says version 1 printed on the pcb.

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Orca 0.42

### Electronics

Creality 4.2.7

### LCD/Controller

MKS H43

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/15754199/Configuration.zip)

shabgarde, 2024-06-10:
> ### Did you test the latest `bugfix-2.1.x` code?
> Yes, and the problem still exists.
>
> ### Bug Description
> I am using a MKS H43 with the creality 4.2.7 board. Everything is working fine. I get the [...]

According to the makerbase team's announcement, this display model is compatible with Marlin 2.0.x . so dont work with marlin 2.1.x

mlueft, 2024-06-10:
hi shapgarde,

thanks for your response, but i am not sure what to think about it.

I have found the following information:
* the makerbas version of marlin was forked from 2.0.x
* mks h43 says it supports all versions of 2.x
* the current marlin documentation says it is supported.

as far as i understand it right h43 should work with version 2.1.x

regards,

shabgarde, 2024-06-12:
> hi shapgarde,
> 
> thanks for your response, but i am not sure what to think about it.
> 
> I have found the following information:
> 
> * the makerbas version of marlin was forked from 2.0.x
> * mks h43 says it supports all versions of 2.x
> * the current marlin documentation says it is supported.
> 
> as far as i understand it right h43 should work with version 2.1.x
> 
> regards,
> 
> p.s.: Today i have discovered, that the display shows the content of the SD-Card when inserted on the creality board. So I am sure the communication is basically working. I am thinking in all directions. The touch module is working too, because i can change pages.

No, you are wrong. This display has an independent processor. So, even if it is not connected to the main board, you can still move between the pages AND the display shows the content of the SD-Card.

If you see bad nozzle temperature and heat on the display. We can say that the display and the main board have established communication

mlueft, 2024-06-12:
no, you are wrong. I know the display has its own processor. but i entered the sd card on the motherboard.
further more it is marlin that says the display to show the main screen after booting and the bootscreen. if i disable the bootscreen in the marlin config the display is not showing it. and the display is showing the bed and extruder temperature.
as far as i understand the marlin source code right. the display send the info about a button event and marlin shows the corresponding screen. that‘s the reason marlin knows the screen and button code of all screens and buttons.

regards,

closed 2024-09-25 by github-actions[bot]