MarlinFirmware/Marlin #27161: [BUG] Extruder auto fan not working properly with MKS TinyBee

danvie4, 2024-06-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Extruder fan (hotend cooling fan) always runs even when the extruder is well below the set temperature. I've tried messing around with it a lot, but I'm not entirely sure if it's a bug or my fault. As it is currently (in the configuration file), the #define E0_AUTO_FAN_PIN -1 line is commented out. My understanding is that it's supposed to use the default pins specified in the pins file. I also tried overriding it and setting it to #define E0_AUTO_FAN_PIN 147 (and 148) but 147 throws an error ("You cannot set E0_AUTO_FAN_PIN equal to FAN0_PIN.") and 148 compiles fine but the fan still always runs.
[Config Files.zip](https://github.com/user-attachments/files/15755698/Config.Files.zip)

### Bug Timeline

I'm not sure

### Expected behavior

I expected the fan to stay off and only turn on when the temperature reached 50c

### Actual behavior

The fan runs all of the time

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Anet et4+

### Electronics

MKS TinyBee Motherboard

### LCD/Controller

MKS12864 V3

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Any help is greatly be appreciated!

ellensp, 2024-06-10:
"E0_AUTO_FAN_PIN -1 line is commented out. My understanding is that it's supposed to use the default pins specified in the pins file"  

The E0_AUTO_FAN_PIN in Marlin/src/pins/esp32/pins_MKS_TINYBEE.h is commented out. So there is no default pin

You cannot set the auto fan to the same pin as the gcode controlled fan. The gcode controlled fan is always FAN0_PIN ie 147
So you cannot set E0_AUTO_FAN_PIN to FAN0_PIN or 147 by design

148 should work though.

danvie4, 2024-06-10:
Thanks! That appears to sort of have worked, but something is still not quite right. I had my fans wired opposite (hot end fan in port 1, part cooling fan in port 2) and it appears that the hot end fan now turns on when it reaches temperature, but now it's the part cooling fan that runs nonstop. M107 doesn't seem to do anything though... Any ideas What my issue might be there?

closed 2024-06-11 by danvie4