MarlinFirmware/Marlin #27163: [BUG] Restart after  MINIMUM_STEPPER_PULSE to MINIMUM_STEPPER_PULSE_NS #27113 PR

vovodroid, 2024-06-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Latest commit [d96c6f8](https://github.com/MarlinFirmware/Marlin/commit/d96c6f867d2174d4cfc560efc65a3666d1cf6ece) fails (Marlin restart) on the first extraction move. So I went to 0169cde (one before  MINIMUM_STEPPER_PULSE to MINIMUM_STEPPER_PULSE_NS #27113 merge, and it work properly.

Does this PR contain something suspicious? 

### Version of Marlin Firmware

bugfix-2.1.x

### Electronics

SKR-3, TMC2209

### LCD/Controller

TFT SPI 2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/user-attachments/files/15775337/Config.zip)

dbuezas, 2024-06-11:
Same here and also a very sluggish UI while steppers move.
Also bisected commits to find the problem in the same commit as this issue reports.
I also have an SKR-3, with TMC2209 drivers.

I have a 20MHz oscilloscope at hand in case measuring pins is of use.

dbuezas, 2024-06-11:
I should mention that the printer also often stops mid print for user confirmation, which I think means the planner is starved.
Maybe a small bug where the steppers are driven at an order of magnitude faster rate than intended?

thisiskeithb, 2024-06-11:
@mh-dm: Mind giving this a review?

mh-dm, 2024-06-13:
First, sorry for the issues. I've tried a few things but I can't reproduce the reported issues.

Looking at #27113 there are basically 2 parts. First the actual `MINIMUM_STEPPER_PULSE_NS` change then there's a refactor to switch quite a few things to `constexpr`. It would help to figure out which part of #27113 is causing the issue.

So could one of you please:
- make sure you are synced right _before_ 27113
- then download the [first commit of #27113](https://github.com/MarlinFirmware/Marlin/pull/27113/commits/ddb8189da75f971e4081b0dfafde85ec5068b334) as a patch:
`wget https://github.com/MarlinFirmware/Marlin/pull/27113/commits/ddb8189da75f971e4081b0dfafde85ec5068b334.patch`
- then apply the patch locally:
`git am ddb8189da75f971e4081b0dfafde85ec5068b334.patch`

and report back whether it's showing the same symptoms or not.

vovodroid, 2024-06-13:
I've tried first commit only. Actually it's the same, just crashes a bit later.

mh-dm, 2024-06-13:
Thank you. This was quite weird but I think I figured out what's going on.

First: I've tried to build with the configuration files attached and I'm getting `#error "NONLINEAR_EXTRUSION doesn't currently support multi-extruder setups.`. I changed a few things to get it to build. But it was a good error to get since `NONLINEAR_EXTRUSION` turns out to be relevant.

So here's what I think is happening.
- You're using SKR 3, which runs at a blistering 480Mhz
- You're using TMC drivers, which can top out at 5 million steps/s and has a minimum pulse duration of 100ns.
- You have `ADAPTIVE_STEP_SMOOTHING` enabled which results in oversampling. This oversampling is calculated based on an estimate of the number of cycles the stepper isr will take to run
- You're using `NONLINEAR_EXTRUSION` which happens to blindly increase oversampling without care for what the hardware can actually do
- PR 27113 fixes `MIN_STEPPER_PULSE_CYCLES` to correctly compute the number of cycles that a pulse of `MINIMUM_STEPPER_PULSE_NS` will take: `constexpr uint32_t min_stepper_pulse_cycles = _min_pulse_high_ns * CYCLES_PER_MICROSECOND / 1000;`

The result of all of this is with PR 27113 on SKR3 w/ TMC the `MIN_STEPPER_PULSE_CYCLES` goes from `960` (wrong) to `48` (correct as that number of cycles at 480Mhz takes 100ns) which means `MIN_STEP_ISR_FREQUENCY` goes from 75519UL to 136518UL. To give some context, assuming 80steps/mm a move at 100mm/s normally runs at 8000steps/s. Due to `ADAPTIVE_STEP_SMOOTHING` it will then be oversampled at 16x (136518/8000 is ~17) *times* 2x oversampling (due to `NONLINEAR_EXTRUSION`). That's a lot of oversampling.

I'm not seeing this issue on LPC176x w/ TMC because that's max 120Mhz and the absolute change in `MIN_STEPPER_PULSE_CYCLES` is much less so it doesn't affect `MIN_STEP_ISR_FREQUENCY` much.

**To confirm whether my analysis is correct and to figure out a fix could you please change this section in `stepper.cpp` from**:

```C++
        // Nonlinear Extrusion needs at least 2x oversampling to permit increase of E step rate
        // Otherwise assume no axis smoothing (via oversampling)
        oversampling_factor = TERN0(NONLINEAR_EXTRUSION, 1);
```

to just

```C++
        oversampling_factor = 0;
```

and report back whether this fixes your issues.

vovodroid, 2024-06-13:
> * You're using SKR 3, which runs at a blistering 480Mhz

Even worse, I have 550Mhz version))

> #error "NONLINEAR_EXTRUSION doesn't currently support multi-extruder setups.

Yes, I just commented out this check in Sanity.h. I use second extruder for support interface, so I don't mind to use values from main extruder for both of them.

> change this section in stepper.cpp

I can change config instead, if you want. Assuming you are right and change in stepper.cpp helps. Can I use it for printing or it's good just for test only?

mh-dm, 2024-06-13:
If it helps it should be good enough for actual printing. I'm preparing that change as a PR.

dbuezas, 2024-06-13:
`oversampling_factor = 0;` solved the issue in my machine

vovodroid, 2024-06-13:
Could you try to add 
```
         #if ENABLED(NONLINEAR_EXTRUSION)
            oversampling_factor = max(oversampling_factor , 1);
         #endif
```
as I suggested?

mh-dm, 2024-06-13:
> Or it anyway will be oversampled as high as possible in while loop?

Yes, it will be sampled as high as possible. There's no need to try to use something like `max(oversampling_factor , 1);`

dbuezas, 2024-06-13:
I'm printing samples to confirm non-linear extrusion still works as before (yes remembered to invert A and B)

dbuezas, 2024-06-13:
yes, still works fine

vovodroid, 2024-06-13:
Great! I hope that full PR with this fix works as well.

PR #27171 (2024-06-14): Don't blindly increase oversampling with `NONLINEAR_EXTRUSION`
### Description

Don't blindly always increase oversampling with `NONLINEAR_EXTRUSION`. Just increasing oversampling can result in a stepper ISR rate higher than the microcontroller can handle, leading to hard to debug behavior.

While `NONLINEAR_EXTRUSION` does benefit from oversampling that does not mean it requires it to function. Let's take some common numbers as an example: 0.2mm layer height, 0.4mm line width, and 100mm/s speed. That's:

- 8000 steps/s for a motion axis (assuming we move just in X or just in Y at 80steps/mm)
 [... cut]
closed 2024-06-14 by thinkyhead