MarlinFirmware/Marlin #27166: [BUG] NONLINEAR_EXTRUSION has no effect

dbuezas, 2024-06-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I measured and extruded at different velocities and came up with M592 A-0.004 B0.00126 C1
I then tried again and it had no effect.
I made the extrusions with `G1 E+90 F125`  up to `G1 E+90 F375`, and basically found an error of 2.5% at 6mm/s.
[image]

To double check I tried using C1.05 and still had no effect.

I of course enabled NONLINEAR_EXTRUSION in configuration_adv.h and sending M592 without parameters reports the values back.

Could the feature be broken?

### Bug Timeline

I didn't try the feature before

### Expected behavior

To measure less error when extruding 100mm

### Actual behavior

NONLINEAR_EXTRUSION has no effect

### Steps to Reproduce

M592 A-0.004 B0.00126 C1.05 

measure before and after

### Version of Marlin Firmware

Latest bugfix 2.1.x

### Printer model

Ultimaker 2

### Electronics

SKR3

### Other add-ons

Orbiter v2 extruder

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Archive.zip](https://github.com/user-attachments/files/15777679/Archive.zip)

vovodroid, 2024-06-11:
Can you try some really huge correction, like C2?

dbuezas, 2024-06-11:
> Can you try some really huge correction, like C2?

Set with `M592 A-0.004 B0.00126 C2`
Corroborated setting with 
```
M592
  M592 A-0.00 B0.00 C2.00 <- response
```

Extruded with 
```
G91
G1 E+90 F125
G90
```

--> Resulted in 90mm extruded. C2 had no effect whatsoever. <--

Now I am wondering if NONLINEAR_EXTRUSION does not affect extrusions without other axes moving?

> N.B.: in latest [bugfix-2.1.x](https://github.com/MarlinFirmware/Marlin) B and A were swapped to match mathematics Av^2+Bc notation.

Good to know 👍 
My branch's parent is d38261c9084a5036e6598d4ab15da3e17447f3ad from June the 8th (before the naming change)

dbuezas, 2024-06-11:
> Now I am wondering if NONLINEAR_EXTRUSION does not affect extrusions without other axes moving?

I think this is it

vovodroid, 2024-06-11:
> Now I am wondering if NONLINEAR_EXTRUSION does not affect extrusions without other axes moving?

No, that was fixed by me))) in https://github.com/MarlinFirmware/Marlin/pull/26824

dbuezas, 2024-06-11:
Oh, I see, so the extrusion is interpreted as an unretract b/c no other axis is moving?

dbuezas, 2024-06-11:
Closed as the docs PR clarify that this was intentional

closed 2024-06-11 by dbuezas

vovodroid, 2024-06-11:
> so the extrusion is interpreted as an unretract b/c no other axis is moving?

Yes.

dbuezas, 2024-06-11:
NONLINEAR_EXTRUSION is working but it made my machine extremely slow. Prints often get interrupted waiting for user input and the UI updates become super slow whenever motors are moving. I have an SKR3 which has a 480MHz mcu so something went very wrong. Is there any caveat I missed?

dbuezas, 2024-06-12:
The issue was another recently merged PR.

Non linear extrusion works like a charm.

I used https://hotend-flow-tester.netlify.app/ to create heaps of plastic that should be equal, but at different extrusion rates, and then measured their weight (mg). (test idea from cnc kitchen https://www.cnckitchen.com/blog/extrusion-system-benchmark-tool-for-fast-prints ) 

Then let google sheets fit a quadratic on `expected/measured`, applied the correction and reprinted. Here's the result:

[image]

vovodroid, 2024-06-13:
I use ruler, it's more accurate and wastes less of filament.

dbuezas, 2024-06-13:
I found weighing more convenient and faster for me, 10 measurements add up to 6 grams and 5 mins of printing so I'm ok with that.

vovodroid, 2024-06-13:
> 10 measurements

Why so much? Actually two are enough - at maximum suggested flow rate and in the middle.

dbuezas, 2024-06-13:
To fit a quadratic and ensure I don't end up under extruding at slow speeds.

vovodroid, 2024-06-13:
> To fit a quadratic

Three points (zero, middle, end) is enough.

> ensure I don't end up under extruding at slow speeds.

You can assume that there is no underextrusion at zero speed (if esteps is correct).

dbuezas, 2024-06-13:
Yes, a quadratic has 3 variables and therefore 3 points are enough to find the constants. Regression is not the same as solving the equation, it's about minimising error.

dbuezas, 2024-06-13:
With my setup it did matter:
[image]
If you pick the "wrong" 3 points, you'd get a completely different curve.