MarlinFirmware/Marlin #27169: [BUG] STM32F103RC_btt: M997 crashes the board instead of resetting it

ali1234, 2024-06-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Sending the command `M997` causes the board to freeze forever.

### Bug Timeline

I don't know, I never tried using M997 before today, so as far as I know it has never worked.

### Expected behavior

`M997` should cause STM32 boards to reset.

### Actual behavior

Sending the command `M997` causes the board to freeze forever.

### Steps to Reproduce

1. Send M997 command to printer.

### Version of Marlin Firmware

2.1.2.3

### Printer model

Creality Ender-3

### Electronics

BTT SKR E3 DIP V1.1

### LCD/Controller

Stock

### Other add-ons

None

### Bed Leveling

None

### Your Slicer

None

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

No modification. The bug happens with example config:

Creality/Ender-3/BigTreeTech SKR E3-DIP V1.1

thisiskeithb, 2024-06-12:
> Don't forget to include
> A ZIP file containing your Configuration.h and Configuration_adv.h.

You forgot to do this. Please attach your `bugfix-2.1.x` configs.

ali1234, 2024-06-12:
> No modification. The bug happens with example config:
> Creality/Ender-3/BigTreeTech SKR E3-DIP V1.1

[configs.zip](https://github.com/user-attachments/files/15811482/configs.zip)

ellensp, 2024-06-13:
This uses NVIC_SystemReset, Failure of reset is almost always due to something external driving the NRST and preventing it getting LOW.

I see you have enabled uart 2, does this means you have a tft plugged in?
if so unplug it, it may be holding the reset line high   

reset is on the following ports: TFT, SWD, EXP1 and SPI1 so for testing please remove anything you have plugged into these ports that could potentially be holding the reset pin high.

ali1234, 2024-06-13:
I have nothing connected to SWD, SPI1, or TFT. I have the stock Ender 3 LCD on EXP1. Unplugging it did not change the result.

After checking the schematic, the board has a reset button which is wired like this:

[image]

Pulled up by 4.7k - could that be the problem?

ali1234, 2024-06-13:
Even pressing that reset button does not reset the board - it makes the status LED light while it is held down, but the board does not boot up when released. Even when not running M997. I would guess that the reset is actually successful, but the firmware is crashing due to some hardware being unexpectedly already initialized. Probably in the vendor bootloader?

thisiskeithb, 2024-06-13:
> The bug happens with example config:
> Creality/Ender-3/BigTreeTech SKR E3-DIP V1.1

Using this config, I set up an SKR E3 DIP V1.1 (RET6 version) with TMC2208s and stock Ender-3 LCD...

- [x] Physical reset button works
- [x] `M997` works
- [x] PIO `STM32F103RC_btt` environment works
- [x] PIO `STM32F103RE_btt` environment works

thisiskeithb, 2024-06-13:
I suspect your board has a hardware fault.

closed 2024-06-13 by thisiskeithb

ali1234, 2024-06-14:
It was actually the SD card. Although it works fine for all operations including reflashing the firmware after power-on-reset, resetting the board by any method while it is inserted causes Marlin to boot loop. The bad SD card is unbranded 128MB and was supplied with the SKR board, and the good SD card is Netac 8GB supplied with the Ender 3.