MarlinFirmware/Marlin #27170: [BUG] (bug summary) G34 probes only the left side of my B1 SE plus(SKR 3) BLtouch duel Z

Jmckenzie84, 2024-06-13:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm Having an issue running the G34 on my B1 plus with duel Z where the my bltouch probe doesn't move to the right side of the printer. the cables are at the are connected to ZAM and E1M and I enabled Z2 driver on the configuration.h file. I'm using Marlin 2.1.X bugfix
[Configuration.zip](https://github.com/user-attachments/files/15812752/Configuration.zip)

### Expected behavior

I expect it to move left to right probing each side when running the G34 command

### Actual behavior

It Probes the left side then it errors and goes back to the center.

### Steps to Reproduce

running the G34 command

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Biqu B1 SE Plus

### Electronics

SKR 3

### Other add-ons

Bltouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-06-13:
I must have made my custom B1 SE Plus build public at some point 😄 

You have both `BLTOUCH` and `PROBE_ENABLE_DISABLE` enabled.

While my current B1 SE Plus build is not a 1:1 match to your config, `G34` works fine on current `bugfix-2.1.x` (e0d683b).

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2024-06-13 by thisiskeithb

Jmckenzie84, 2024-06-13:
I recompiled my firmware with PROBE_ENABLE_DISABLE disabled  and ran G34 and still get

Recv: G34 Iteration: 1
[...]
Recv: Probing Failed
Recv: G34 aborted.

thisiskeithb, 2024-06-13:
Please use the resources linked above for help with configuration and troubleshooting.