MarlinFirmware/Marlin #27176: [BUG] (bug 82 |   #define GET_TEXT(MSG) GET_LANG(LCD_LANGUAGE)::MSG)

cyborgtr1977, 2024-06-14:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

In file included from Marlin\src\module\../inc/../core/language.h:597,
                 from Marlin\src\module\../inc/MarlinConfig.h:56,
                 from Marlin\src\module\motion.h:31,
                 from Marlin\src\module\motion.cpp:27:
Marlin\src\module\motion.cpp: In function 'bool homing_needed_error(main_axes_bits_t)':
Marlin\src\module\motion.cpp:1491:31: warning: format '%s' expects a matching 'char*' argument [-Wformat=]
 1491 |       sprintf_P(msg, GET_TEXT(MSG_HOME_FIRST), need);
Marlin\src\module\../inc/../core/multi_language.h:82:49: note: in definition of macro 'GET_TEXT'
   82 |   #define GET_TEXT(MSG) GET_LANG(LCD_LANGUAGE)::MSG
      |                                                 ^~~
Marlin\src\module\motion.cpp:1491:31: warning: format '%s' expects a matching 'char*' argument [-Wformat=]
 1491 |       sprintf_P(msg, GET_TEXT(MSG_HOME_FIRST), need);
Marlin\src\module\../inc/../core/multi_language.h:82:49: note: in definition of macro 'GET_TEXT'
   82 |   #define GET_TEXT(MSG) GET_LANG(LCD_LANGUAGE)::MSG

### Version of Marlin Firmware

2.1.2.3

### Printer model

PRUSA KLON

### Electronics

SKR 1.4 TMC 2209 

### LCD/Controller

MKS MİNİ 12864V3

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-06-14:
> Did you test the latest bugfix-2.1.x code?
> Yes, and the problem still exists.

> Marlin\src\module\motion.cpp:1491:31: warning: format '%s' expects a matching 'char*' argument [-Wformat=]
> 1491 | sprintf_P(msg, GET_TEXT(MSG_HOME_FIRST), need);

"`sprintf_P(msg, GET_TEXT(MSG_HOME_FIRST), need);`" doesn't exist in `bugfix-2.1.x`.

Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

ellensp, 2024-06-15:
Cannot reproduce this Warning
Code is not present in bugfix 
No Configuration files provided to reproduce issue.

Nothing can be done here

closed 2024-06-15 by ellensp