MarlinFirmware/Marlin #27181: M306 T produces negative responsiveness value

Drzet, 2024-06-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Having installed a new, original E3D Revo 12v hotend I ran M306 and got the below values:

`:Recv: MPC Autotune finished! Put the constants below into Configuration.h
Recv: MPC_BLOCK_HEAT_CAPACITY 9.36
Recv: **MPC_SENSOR_RESPONSIVENESS -0.9229**
Recv: MPC_AMBIENT_XFER_COEFF 0.0525
Recv: MPC_AMBIENT_XFER_COEFF_FAN255 0.0884`

The negative responsiveness value led to a thermal runaway issue until I realized what was happening and removed the minus sign.

### Bug Timeline

With a new E3D Revo.

### Expected behavior

Normal values

### Actual behavior

Recv: **MPC_SENSOR_RESPONSIVENESS -0.9229**

### Steps to Reproduce

T1 (it's a dual system)
M306 T

### Version of Marlin Firmware

2.1.2.3

### Printer model

Custom coreXY Hypercube

### Electronics

SKR 3

### LCD/Controller

BTT-12864

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/15858901/Marlin.zip)

rondlh, 2024-06-30:
Do you have a ceramic hotend heater, if so this could be related to #27223

Did you correctly define the MPC_HEATER_POWER? 
#define MPC_HEATER_POWER { 40.0f, 40.0f }

You have defined 2 different types of thermistors for the hotends, make sure it's correct.
#define TEMP_SENSOR_0 1010
#define TEMP_SENSOR_1 5

Drzet, 2024-06-30:
> Do you have a ceramic hotend heater
It's e3d Revo, I believe it's ceramic.

> Did you correctly define the MPC_HEATER_POWER?
40W, as per published e3d specs

>You have defined 2 different types of thermistors
Yes, the second hotend is the old V6

rondlh, 2024-07-08:
> > Do you have a ceramic hotend heater
> 
> It's e3d Revo, I believe it's ceramic.
> 
> > Did you correctly define the MPC_HEATER_POWER?
> 
> 40W, as per published e3d specs
> 
> > You have defined 2 different types of thermistors
> 
> Yes, the other hotend is the old V6 with pt1000

OK, good to know. 
Here you can see the heater characteristics:
https://e3d-beta.myshopify.com/pages/heatercore-range-datasheet
[image]

I think you should look at the orange line, which shows that if you print at 225 degrees C, that the power goes down to about 28W, you could try that number instead of the 40W.

Drzet, 2024-07-08:
here's the result @28W
Recv: MPC Autotune finished! Put the constants below into Configuration.h
Recv: MPC_BLOCK_HEAT_CAPACITY 7.49
Recv: MPC_SENSOR_RESPONSIVENESS **-1.6147**
Recv: MPC_AMBIENT_XFER_COEFF 0.0376
Recv: MPC_AMBIENT_XFER_COEFF_FAN255 0.0699

rondlh, 2024-07-09:
All numbers reported seem reasonable, except this minus sign. Are you sure the thermistor type is correct?
Do the readings at room temperature make sense? (Should be about within 1 or 2 degrees of what your bed is reporting).

I'm not sure how you set the new heater power, best way to test is just to issue a M306 P28 before the M306 T. If you change the value in the configuration.h, the new value will not be used immediately... Run an M306 without parameters to see the currently active MPC values.

Drzet, 2024-07-09:
> Are you sure the thermistor type is correct?

Type 5, as per specifications

> Do the readings at room temperature make sense?

B:23 T0:24 T1:23

> Run an M306 without parameters

M306 E1 P28.00 C9.36 R0.1810 A0.0525 F0.0884 H0.0036
I changed the R value manually because the negative one was causing thermal runaways.

tombrazier, 2024-07-10:
Revo hotends (and PTC hotends generally) break an assumption of the _asymptotic_ MPC autotuning method and this results in nonsense negative sensor responsiveness values.

@StevilKnevil added the _differential_ autotuning method which, I believe, was merged in #25503 (but originally submitted as #25517). This method is slightly less accurate but should give sensible results for all hotend types.

Since #25503 `M306 T` with no `S` parameter should attempt the _asymptotic_ method and if that gives a negative sensor responsiveness it then should use the _differential_ method. This bug report perhaps suggests that this default behaviour might not be working properly.

@StevilKnevil any thoughts?

Drzet, 2024-07-10:
> @StevilKnevil added the differential autotuning method...

Yes, I discovered that in Marlin documentation and tried it with the explicit `M306 TD` command. Got something like -5.5 responsiveness

tombrazier, 2024-07-10:
You need `M306 T S1` to do a straight differential tune.

Drzet, 2024-07-10:
> Automatic Differential Tuning
If asymptotic tuning (above) fails, or Differential Tuning is forced with M306 TD then automatic tuning uses the following algorithm.

That's from [MPC docs](https://marlinfw.org/docs/features/model_predictive_control.html)
I'll try S1 later

tombrazier, 2024-07-10:
Hmm, looks like the "Automatic Differential Tuning" part of the docs has a few errors. I also see some math values that don't have enough $ signs and are not rendered correctly on my web browser.

But also, the next section in the doc explains why you are still getting negative sensor responsiveness. The value calculated by the differential tuning gets thrown away and replaced with the value from the asymptotic tune. Reading the source code, it does appear to have the described behaviour for the Auto Select tuning case, i.e. `M306 T` or `M306 T S0`. When differential tuning is selected explicitly with `M306 T S1` I think the differential tuning result will be retained.

Drzet, 2024-07-13:
Send: M306 T S1
Recv: MPC Autotune start for E0
Recv: Active Extruder: 0
[...]
Recv: Active Extruder: 0
Recv: X:106.00 Y:110.00 Z:10.00 E:121.71 Count A:52160B:-18880 Z:4000
[...]
Recv: Cooling to ambient
[...]
Recv: Heating to over 200C
[...]
Recv: Measuring ambient heatloss at 209.15
[...]
Recv: MPC Autotune finished! Put the constants below into Configuration.h
Recv: MPC_BLOCK_HEAT_CAPACITY 7.45
Recv: MPC_SENSOR_RESPONSIVENESS -2.4951
Recv: MPC_AMBIENT_XFER_COEFF 0.0427
Recv: MPC_AMBIENT_XFER_COEFF_FAN255 0.0699
[...]
Recv: ok P15 B3

tombrazier, 2024-07-14:
Could you include the text that was replaced by `[...]`?

Also: the values this tune gave are all about 80% of the previous values. Did you use the same power value for both tunes? Marlin has no way of knowing whether the power value is correct so it accepts it as given when tuning and bases all other values on it.

Drzet, 2024-07-14:
I have attached logs of M306 T and M306 T S1. Both done with freshly complied latest Marlin 2.1.2.4, right after boot.
[E0_M306_T.txt](https://github.com/user-attachments/files/16216609/E0_M306_T.txt)
[E0_M306_T_S1.txt](https://github.com/user-attachments/files/16216610/E0_M306_T_S1.txt)

thisiskeithb, 2024-07-14:
> Both done with freshly complied latest Marlin 2.1.2.4

Testing should be done with [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip), not a tagged release since releases are actually more than a year behind current code.

Drzet, 2024-07-14:
M306 T, M306 T S1 and M306 TD.
Marlin BugFix 02010300
Each tuning done after powering off and on. Same extruder, same power setting of 28W

[BUGFIX_E0_M306_T.txt](https://github.com/user-attachments/files/16227676/BUGFIX_E0_M306_T.txt)
[BUGFIX_E0_M306_T_S1.txt](https://github.com/user-attachments/files/16227677/BUGFIX_E0_M306_T_S1.txt)
[E0_M306_TD.txt](https://github.com/user-attachments/files/16227678/E0_M306_TD.txt)

tombrazier, 2024-07-14:
Okay, so...

- `M306 TD` is incorrect and the documentation needs correcting.
- `M306 S1` works and gives very plausible results (for comparison my CHC hotend has MPC_BLOCK_HEAT_CAPACITY 5.54, MPC_SENSOR_RESPONSIVENESS 0.229, MPC_AMBIENT_XFER_COEFF 0.056 and MPC_AMBIENT_XFER_COEFF_FAN255 0.07 which are all very much in the same ballpark as the `S1` results for a fairly similar hotend).
- Using the automatic method, i.e. no `S` parameter to `M306`, the fallback to differential is incorrectly overridden at the last minute in the "Refining estimates" step. If the automatic method has established that the asymptotic method is incorrect then the refinement step is also incorrect and should not happen.

I can fix both code and docs pretty easily.

Drzet, 2024-07-14:
While at it, can you also fix `#if defined(NEOPIXEL_TYPE) && NEOPIXEL_TYPE != NEO_RGB
  #error "Your FYSETC/MKS/BTT/BEEZ Mini Panel requires NEOPIXEL_TYPE to be NEO_RGB."`  ?
These come in all sorts, BRG, GBR...
Allegedly fixed as mentioned [here](https://github.com/MarlinFirmware/Marlin/issues/24116) but it actually isn't

thisiskeithb, 2024-07-14:
> These come in all sorts, mine is BRG

They are all RGB. The fact that you have to set it as BRG is a workaround for a bug on certain motherboards / processors:  

- https://github.com/MarlinFirmware/Marlin/issues/26260

tombrazier, 2024-07-15:
It turns out there are several other bugs with the S1 tuning. Fixes for all of them coming soon to a PR near you.

closed 2024-08-05 by thinkyhead