MarlinFirmware/Marlin #27182: [BUG] (bug summary) MKS SGen L v2.0/ MKS TS35 v2.0

DragonLord1981, 2024-06-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

In file included from Marlin\src\HAL\LPC1768\tft\tft_spi.cpp:29:
Marlin\src\HAL\LPC1768\tft\tft_spi.h:38:4: error: #error "Invalid TFT SPI configuration."
   38 |   #error "Invalid TFT SPI configuration."

### Bug Timeline

New for me

### Version of Marlin Firmware

CONFIGURATION_H_VERSION 02010203

### Printer model

Xvico

### Electronics

MKS SGen L v2.0

### LCD/Controller

MKS TS35 v2.0

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-06-16:
Using the latest [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip), replace `/Marlin/Marlin/src/pins/lpc1769/pins_MKS_SGEN_L_V2.h` with this file:
[pins_MKS_SGEN_L_V2.h.zip](https://github.com/user-attachments/files/15858919/pins_MKS_SGEN_L_V2.h.zip)

It compiles, but I haven't tested it on real hardware.

DragonLord1981, 2024-06-16:
Perfect.. Thanks.. i will see if it works on the hardware  :D 

> Using the latest [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip), replace `/Marlin/Marlin/src/pins/lpc1769/pins_MKS_SGEN_L_V2.h` with this file: [...]

Perfect.. Thanks.. i will see if it works on the hardware  :D

DragonLord1981, 2024-06-16:
> Using the latest [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip), replace `/Marlin/Marlin/src/pins/lpc1769/pins_MKS_SGEN_L_V2.h` with this file: [...]

Did kinda work bur got "err - eeprom"
And the screen is black no background light

thisiskeithb, 2024-06-16:
> Did kinda work bur got "err - eeprom"

That's expected when you migrate major versions of Marlin and means you probably weren't running the latest `bugfix-2.1.x` before flashing.

> And the screen is black no background light

I copied the pin layout from another pins file hoping EXP numbering would match up, so I'll have to look at this more closely when I wrap up some other work and can test on real hardware.

ellensp, 2024-06-17:
#define TOUCH_BUTTONS_HW_SPI_DEVICE      1  should be 2

but other than that... it looks ok.

The back light not working doesn't make sense its just an IO pin, are you sure you plugged display exp1 to board exp1 and display exp2 to board exp2 correctly and not switched over?

thisiskeithb, 2024-06-17:
See https://github.com/MarlinFirmware/Marlin/pull/27183 for the verified fix. There's still a quark with a BTT TFT35 SPI display on this motherboard, but I'll look into that later since it'll be a rare combination.

DragonLord1981, 2024-06-17:
> TOUCH_BUTTONS_HW_SPI_DEVICE

Where do i find that setting ? and yes. the machin was working before update

ellensp, 2024-06-17:
In the originally provided updated pins_MKS_SGEN_L_V2.h.zip, it has been updated in  https://github.com/MarlinFirmware/Marlin/pull/27183

DragonLord1981, 2024-06-17:
How do i download that file ? haha sorry.. i tried all the ways that i know

thisiskeithb, 2024-06-17:
> How do i download that file ? haha sorry.. i tried all the ways that i know

Click on the download icon:

[image]

I've also attached it here:

[pins_MKS_SGEN_L_V2.h.zip](https://github.com/user-attachments/files/15875414/pins_MKS_SGEN_L_V2.h.zip)