MarlinFirmware/Marlin #27187: BigTreeTech octopus PRO V1.1 H723 can not compile[BUG] (bug summary)

ishmel2000, 2024-06-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When I try to compile the marlin from the BTT repository, it gives me a fail with this message:

Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:simulator_windows] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F103CB_malyan] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:simulator_windows] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F103CB_malyan] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Resolving STM32H723Zx_btt dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Warning! `src_filter` configuration option in section [env:include_tree] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:simulator_windows] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:STM32F103CB_malyan] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Processing STM32H723Zx_btt (platform: ststm32@~14.1.0; board: marlin_STM32H723Zx; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_STM32H723Zx.html
PLATFORM: ST STM32 (14.1.0) > STM32H723ZE (564k RAM. 512k Flash)
HARDWARE: STM32H723ZET6 400MHz, 564KB RAM, 512KB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 @ 0.0.0 
 - framework-cmsis @ 2.50700.210515 (5.7.0) 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
[... 26 lines cut ...]
==================== 1 failed, 0 succeeded in 00:00:08.739 ====================

### Version of Marlin Firmware

2.0.9.3

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin-bugfix-2.0.9.3.x.zip](https://github.com/user-attachments/files/15889695/Marlin-bugfix-2.0.9.3.x.zip)

ellensp, 2024-06-18:
This code was not supplied by marlin, this is a from the btt marlin fork https://github.com/bigtreetech/BIGTREETECH-OCTOPUS-V1.0/tree/master/Firmware/Marlin-bugfix-2.0.9.3.x , they need to update their fork.

ellensp, 2024-06-18:
They need to apply this https://github.com/MarlinFirmware/Marlin/commit/8fd6f9a2b029caa9fa30c61c81e01236d6bce0f9

ishmel2000, 2024-06-18:
Thanks for the answer. If I understood you well. Those changes need to be applied for the respective files? sorry if I look stupid but this is a new problem for me jajajaj

ishmel2000, 2024-06-18:
**I tried te correction but this is the new problem:**

Marlin\src\HAL\STM32\sdio.cpp: In function 'void HAL_SD_MspInit(SD_HandleTypeDef*)':
Marlin\src\HAL\STM32\sdio.cpp:95:28: error: 'PinMap_SD' was not declared in this scope; did you mean 'PinMap_USB'?
   95 |       pinmap_pinout(PC_9,  PinMap_SD);
      |                            ^~~~~~~~~
      |                            PinMap_USB
*** [.pio\build\STM32H723Zx_btt\src\src\HAL\STM32\sdio.cpp.o] Error 1

**If I correct the problem it gives me this one:**

Compiling .pio\build\STM32H723Zx_btt\FrameworkArduino\wiring_time.c.o
Archiving .pio\build\STM32H723Zx_btt\libFrameworkArduino.a
Linking .pio\build\STM32H723Zx_btt\firmware.elf
c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/STM32H723Zx_btt/src/src/HAL/STM32/MarlinSerial.cpp.o: in function `EmergencyParser::update(EmergencyParser::State&, unsigned char)':
C:\Users\Ismael\Desktop\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.3.x_edit/Marlin\src\HAL\STM32\../../inc/../HAL/./STM32/../../feature/e_parser.h:204: undefined reference to `HostUI::handle_response(unsigned char)'
c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Ismael\Desktop\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.3.x_edit/Marlin\src\HAL\STM32\../../inc/../HAL/./STM32/../../feature/e_parser.h:204: undefined reference to `EmergencyParser::M876_reason'
c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Ismael\Desktop\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.3.x_edit/Marlin\src\HAL\STM32\../../inc/../HAL/./STM32/../../feature/e_parser.h:204: undefined reference to `EmergencyParser::enabled'
c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Ismael\Desktop\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.3.x_edit/Marlin\src\HAL\STM32\../../inc/../HAL/./STM32/../../feature/e_parser.h:204: undefined reference to `EmergencyParser::killed_by_M112'
c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Ismael\Desktop\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.3.x_edit/Marlin\src\HAL\STM32\../../inc/../HAL/./STM32/../../feature/e_parser.h:204: undefined reference to `EmergencyParser::quickstop_by_M410'
c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/STM32H723Zx_btt/src/src/HAL/STM32/usb_serial.cpp.o: in function `EmergencyParser::update(EmergencyParser::State&, unsigned char)':
C:\Users\Ismael\Desktop\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.3.x_edit/Marlin\src\HAL\STM32\../../feature/e_parser.h:204: undefined reference to `HostUI::handle_response(unsigned char)'c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/STM32H723Zx_btt/src/src/HAL/STM32/usb_serial.cpp.o: in function `USBD_CDC_Receive_hook':
C:\Users\Ismael\Desktop\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.3.x_edit/Marlin\src\HAL\STM32/usb_serial.cpp:41: undefined reference to `EmergencyParser::M876_reason'
c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Ismael\Desktop\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.3.x_edit/Marlin\src\HAL\STM32/usb_serial.cpp:41: undefined reference to `EmergencyParser::killed_by_M112'
c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Ismael\Desktop\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.3.x_edit/Marlin\src\HAL\STM32/usb_serial.cpp:41: undefined reference to `EmergencyParser::quickstop_by_M410'
c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Ismael\Desktop\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.3.x_edit/Marlin\src\HAL\STM32/usb_serial.cpp:41: undefined reference to `EmergencyParser::enabled'
c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/STM32H723Zx_btt/src/src/MarlinCore.cpp.o: in function `printingIsPaused()':
C:\Users\Ismael\Desktop\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.3.x_edit/Marlin\src/MarlinCore.cpp:339: undefined reference to `CardReader::flag'
c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Ismael\Desktop\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.3.x_edit/Marlin\src/MarlinCore.cpp:339: undefined reference to `CardReader::file'
c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/STM32H723Zx_btt/src/src/MarlinCore.cpp.o: in function `kill(__FlashStringHelper const*, __FlashStringHelper const*, bool)':
[... cut ...] t*)'
c:/users/ismael/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Ismael\Desktop\BIGTREETECH-OCTOPUS-V1.0-master\Firmware\Marlin-bugfix-2.0.9.3.x_edit/Marlin\src\module\../inc/../HAL/./STM32/../../feature/host_actions.h:92: undefined reference to `HostUI::notify_P(char const*)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32H723Zx_btt\firmware.elf] Error 1

Please I do not know what to do next

thisiskeithb, 2024-06-18:
@ishmel2000: Use Marlin from [here (this repo) ](https://github.com/MarlinFirmware/Marlin) instead of BTT's fork.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server