MarlinFirmware/Marlin #27189: [BUG] compilation for SKR E3 V3 mini

Mesgame1, 2024-06-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I can't compile my firmware with my configurations, I even tried with the basic configuration

### Version of Marlin Firmware

Marlin-BUGFIX-2.1.X

### Printer model

Creality Ender-3

### Electronics

Bigtreetech SKR E3 V3 mini

### LCD/Controller

Bigtreetech TFT 35

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

config Files:  
 [Marlin.zip](https://github.com/user-attachments/files/15892644/Marlin.zip)
 
 

ERROR:
 
 ```*  Executing task: C:\Users\devie\.platformio\penv\Scripts\platformio.exe run 

Processing STM32G0B1RE_btt (platform: ststm32@17.1.0; board: marlin_STM32G0B1RE; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
In file included from c:\marlin-bugfix-2.1.x\marlin\src\inc\MarlinConfigPre.h:39,
                 from c:\marlin-bugfix-2.1.x\marlin\src\inc\marlinconfig.h:28,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
c:\marlin-bugfix-2.1.x\marlin\configuration.h:284:11: error: missing binary operator before token "("
  284 | #if EITHER(PARKING_EXTRUDER, MAGNETIC_PARKING_EXTRUDER)
      |           ^
c:\marlin-bugfix-2.1.x\marlin\configuration.h:986:11: error: missing binary operator before token "("
  986 | #if EITHER(MORGAN_SCARA, MP_SCARA)
      |           ^
c:\marlin-bugfix-2.1.x\marlin\configuration.h:1787:11: error: missing binary operator before token "("
 1787 | #if EITHER(MIN_SOFTWARE_ENDSTOPS, MAX_SOFTWARE_ENDSTOPS)
      |           ^
c:\marlin-bugfix-2.1.x\marlin\configuration.h:1989:11: error: missing binary operator before token "("
 1989 | #if EITHER(AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_BILINEAR)
      |           ^
c:\marlin-bugfix-2.1.x\marlin\configuration.h:3035:11: error: missing binary operator before token "("
[... 185 lines cut ...]
  +<src/module/tool_change.cpp>
  +<src/module/stepper/indirection.cpp>

#
# Default values apply to all 'env:' prefixed environments
#
[env]
framework         = arduino
extra_scripts     = ${common.extra_scripts}
build_flags       = ${common.build_flags}
lib_deps          = ${common.lib_deps}
monitor_speed     = 250000
monitor_eol       = LF
monitor_echo      = yes
monitor_filters   = colorize, time, send_on_enter

#
# Just print the dependency tree
#
[env:include_tree]
platform         = atmelavr
board            = megaatmega2560
build_flags      = -c -H -std=gnu++11 -Wall -Os -D__MARLIN_FIRMWARE__
build_src_filter = +<src/MarlinCore.cpp>
```

thisiskeithb, 2024-06-18:
You're not using the latest `bugfix-2.1.x` and/or have copied an old config in since `BOTH()` and `EITHER()` macros were removed over a year ago.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

closed 2024-06-18 by thisiskeithb

Mesgame1, 2024-06-18:
Hello, how can you explain that even with a stock configuration of the latest version of marlin available on your site. Taking the config and the marlin that you indicate, I cannot compile? You should know that the files and error that I gave you, I already managed to compile them 3 months ago and that it works

thisiskeithb, 2024-06-19:
> Did you test the latest bugfix-2.1.x code?
> Yes, and the problem still exists.

> Version of Marlin Firmware
> Marlin-BUGFIX-2.1.X

The configs you attached have:
`#define CONFIGURATION_H_VERSION 02010202`

This is not the latest `bugfix-2.1.x`. You cannot mix & match configs/Marlin version.

Neither Marlin `bugfix-2.1.x` from https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.1.x or `bugfix-2.1.x` Configurations from https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.1.x contain these macros.

You will need to manually transfer your old config settings to the latest `bugfix-2.1.x` version or use the latest `bugfix-2.1.x` configs from the same config directory you have in your config (`#define CONFIG_EXAMPLES_DIR "config/examples/Creality/Ender-3/BigTreeTech SKR Mini E3 3.0"`)

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server