MarlinFirmware/Marlin #27205: [FR] (Homing Motor Current)

landodragon141, 2024-06-24:
### Is your feature request related to a problem? Please describe.

In the Slicer I can use G-Code to reduce motor current for Sensorless Homing (TMC2209). However when doing UBL or any other Firmware based homing it uses the default current which is not reliable and either tries to overrun the endstop or it detects home in the middle of travel.

### Describe the feature you want

I would like an additional parameter that when a G28 is called it looks to see if there is a "Homing Current" and sets that if different from standard current. This may only be applicable to Sensorless homing but could be useful as a safety feature on high power machines in case of endstop failure.

thisiskeithb, 2024-06-24:
You can find `*_CURRENT_HOME` settings for each axis in `Configuration_adv.h`. These values are used for homing.

Ex:

https://github.com/MarlinFirmware/Marlin/blob/60d843b540fdc886767180e62c4a1f0d33a27f3e/Marlin/Configuration_adv.h#L2974

https://github.com/MarlinFirmware/Marlin/blob/60d843b540fdc886767180e62c4a1f0d33a27f3e/Marlin/Configuration_adv.h#L2994

https://github.com/MarlinFirmware/Marlin/blob/60d843b540fdc886767180e62c4a1f0d33a27f3e/Marlin/Configuration_adv.h#L3014

etc...

closed 2024-06-24 by thisiskeithb

landodragon141, 2024-06-24:
Wow I can't believe I missed that.... Thanks!