MarlinFirmware/Marlin #27206: [BUG] G30 - Single Z-Probe doesn't respect HOME_AFTER_DEACTIVATE

KevWal, 2024-06-25:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

The command "G30" - Single Z-Probe doesn't appeear to respect the HOME_AFTER_DEACTIVATE setting.

I have NO_MOTION_BEFORE_HOMING and HOME_AFTER_DEACTIVATE set in my config file, they work fine normally, enforcing the homing before motion, but then I created some Macros (in Pronterface) using G30 and they just move, even after a deactivation, without a homing.

### Bug Timeline

Noticed 24/06/2024

### Expected behavior

Expected not to move with an error, or to run a G28 automatically

### Actual behavior

It just moves, even if it means it tries to take the axis beyond the maximum limits of the printer (I have no max physical end stops), causing motors to stall and possible damage.  

### Steps to Reproduce

Deactivate the motors
Run a G30 command

### Version of Marlin Firmware

2.1.2.4

### Printer model

Prusa I3

### Electronics

RAMPS 1.4

### LCD/Controller

Panelolu2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/15966293/Configuration.zip)

ellensp, 2024-06-26:
Please test the [`bugfix-2.1.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thisiskeithb, 2024-07-13:
@DerAndere1 opened up a PR that may fix this:

- #27229

Please give that PR a try and report your findings there.

closed 2026-10-30 by thinkyhead