MarlinFirmware/Marlin #27213: [BUG] (bug summary) ZONESTAR_12864OLED No Longer Works in 2.1.x BUGFIX But Works in 2.0.7 Branch

jutochoppa, 2024-06-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I'm migrating to marlin 2.1.x bugfix branch from 2.0.7 lts and for some reason the display is not working for my Z6 ZM3E2 Board. 
The display is black as if it has no power.

I have configured it successfully so  everything else works. I just finished a print with it too.
If anyone could assist with this, I'd greatly appreciate the help. I've spent hours trying to figure this out.

### Expected behavior

Display to work/ show life

### Version of Marlin Firmware

2.1.x BUGFIX

### Printer model

ZONESTAR Z6,  ZM3E2 motherboard

### LCD/Controller

ZONESTAR_12864OLED

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/15997288/Configuration.zip)
[Configuration_adv.zip](https://github.com/user-attachments/files/15997289/Configuration_adv.zip)

ellensp, 2024-06-27:
the lcd pin out matches what is in https://github.com/ZONESTAR3D/source-code-for-3d-printer/blob/main/Marlin/src/pins/stm32f1/pins_ZM3E2_V1_0.h

jutochoppa, 2024-06-27:
> the lcd pin out matches what is in https://github.com/ZONESTAR3D/source-code-for-3d-printer/blob/main/Marlin/src/pins/stm32f1/pins_ZM3E2_V1_0.h So its probably not a simple pins issue
> 
> Please confirm this is your motherboard https://ae01.alicdn.com/kf/Sba938ab4ed7f451b9c5613b1295cc3d5F/New-Arrival-32-bits-All-In-One-3D-Printer-Laser-engraving-Control-Board-5-Stepping-Motor.jpg_.webp
> 
> The pins also seem to match the silk screen on this image.

That is indeed the motherboard.
Here is a copy of my latest build log, If it matters: https://pastebin.com/j27MkkKb

And some layman detective work:

**From Zonestars provided source, inside ultralcd_DOGM.h, starting at line 140:** 

#elif EITHER(MKS_12864OLED, ZONESTAR_12864OLED)
  // MKS 128x64 (SH1106) OLED I2C LCD
  // - or -
  // Zonestar SH1106 OLED SPI LCD
  #define FORCE_SOFT_SPI
  #if ENABLED(ALTERNATIVE_LCD)
    #define U8G_CLASS U8GLIB_SH1106_128X64_2X                   // 4 stripes
  #else
    #ifdef __STM32F1__
	#define U8G_CLASS U8GLIB_SH1106_128X64_SW_SPI_HAL
	#else
    #define U8G_CLASS U8GLIB_SH1106_128X64                      // 8 stripes
    #endif    
  #endif

**"U8GLIB_SH1106_128X64_SW_SPI_HAL" Defined in HAL_LCD_class_defines.h in Zonestars source:**

#ifdef U8GLIB_SH1106
extern u8g_dev_t u8g_dev_sh1106_128x64_HAL_sw_spi;
class U8GLIB_SH1106_128X64_SW_SPI_HAL : public U8GLIB {
public:
  U8GLIB_SH1106_128X64_SW_SPI_HAL(pin_t sck, pin_t mosi, pin_t cs, pin_t reset = U8G_PIN_NONE)
  : U8GLIB(&u8g_dev_sh1106_128x64_HAL_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, U8G_PIN_NONE, (uint8_t)reset)
  {  }
};
#endif

jutochoppa, 2024-06-27:
Just to note, I've tried  marlinfirmware/U8glib-HAL@0.4. 5 to version 0.5.4

ellensp, 2024-06-27:
I think your on to something here.
That code was definitely added changing the default behavior of a ZONESTAR_12864OLED_SSD1306 when on a stm32  (which your board is)

jutochoppa, 2024-06-27:
I have updated the HAL_LCD_class_defines.h to include the updated functions
`#ifdef U8GLIB_SH1106 //Fixed
extern u8g_dev_t u8g_dev_sh1106_128x64_HAL_sw_spi;
class U8GLIB_SH1106_128X64_SW_SPI_HAL : public U8GLIB {
public:
  U8GLIB_SH1106_128X64_SW_SPI_HAL() : U8GLIB() { }
  U8GLIB_SH1106_128X64_SW_SPI_HAL(pin_t sck, pin_t mosi, pin_t cs, pin_t reset = U8G_PIN_NONE) {init(sck, mosi, cs, reset); }
  void init(pin_t sck, pin_t mosi, pin_t cs, pin_t reset=U8G_PIN_NONE) {
    U8GLIB::init(&u8g_dev_sh1106_128x64_HAL_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, U8G_PIN_NONE, (uint8_t)reset);
  }
};
#endif

#ifdef U8GLIB_SSD1306 //Fixed
extern u8g_dev_t u8g_dev_ssd1306_128x64_HAL_sw_spi;
class U8GLIB_SSD1306_128X64_SW_SPI_HAL : public U8GLIB {
public:
  U8GLIB_SSD1306_128X64_SW_SPI_HAL() : U8GLIB() { }
  U8GLIB_SSD1306_128X64_SW_SPI_HAL(pin_t sck, pin_t mosi, pin_t cs, pin_t reset = U8G_PIN_NONE) {init(sck, mosi, cs, reset); }
  void init(pin_t sck, pin_t mosi, pin_t cs, pin_t reset=U8G_PIN_NONE) {
    U8GLIB::init(&u8g_dev_ssd1306_128x64_HAL_sw_spi, (uint8_t)sck, (uint8_t)mosi, (uint8_t)cs, U8G_PIN_NONE, (uint8_t)reset);
  }
};
#endif`

I've also included u8g_dev_ssd1306_sh1106_128x64_SWSPI.cpp from Zonestars repo to /Marlin/src/lcd/dogm/u8g/u8g_dev_ssd1306_sh1106_128x64_SWSPI.cpp

The display now works. there's a little glitch on the right side but it might be because its slightly cracked.

Please add this fix to the repo.

ellensp, 2024-06-28:
This 'fix' as is would break all other i2c SH1106 displays on any stm32 controllers so it needs to be made to play nice for everyone

ellensp, 2024-06-28:
@jutochoppa please test https://github.com/ellensp/Marlin/tree/zonestar-spi-oleds 
these are the change I made https://github.com/MarlinFirmware/Marlin/compare/bugfix-2.1.x...ellensp:Marlin:zonestar-spi-oleds

ellensp, 2024-06-28:
Pr created so everyone gets the fix

jutochoppa, 2024-06-29:
> This 'fix' as is would break all other i2c SH1106 displays on any stm32 controllers so it needs to be made to play nice for everyone

Sorry. I'm a bit of an under-qualified dummy at this. I barley know how to use GitHub.

To get your fix working (which itself does)  /// Marlin 2.1.x itself, I also had to

1. Add thermistor_56.h and thermistor_57.h
2. Update thermistors.h
3. Add ZONESTAR_ZM3E2.py to /buildroot/share/PlatformIO/scripts/ZONESTAR_ZM3E2.py
4. Add ZONESTAR_ZM3E2.ld to buildroot/share/PlatformIO/ldscripts
5.  Add build_flags.py to Marlin/src/HAL/STM32F1/
6.  Modify stepper.cpp

Again, I am a professional dummy, so there are probably better solutions to the modifications made.
The rest of Zonestar's scripts are in their repo.

[Zonestar Z6 fix.zip](https://github.com/user-attachments/files/16038199/Zonestar.Z6.fix.zip)

thisiskeithb, 2024-06-29:
While the LCD issue was a bug, the rest of this sounds like a [Feature Request](https://github.com/MarlinFirmware/Marlin/issues/new/choose) to port ZONESTAR Z6 / ZM3E2 support upstream and should be requested separately.

jutochoppa, 2024-06-29:
Okay, thank you.

closed 2024-07-05 by thinkyhead