MarlinFirmware/Marlin #27223: [FR] Improved MPC support for ceramic hotend heaters (with temperature dependent heating power).

rondlh, 2024-06-29:
### Is your feature request related to a problem? Please describe.

Yes, MPC doesn't always work well for ceramic hotend heaters

### Are you looking for hardware support?

No

### Describe the feature you want

MPC (Predictive Model temperature control) generally works very well, especially with the common 6x20mm hotend heater cartridges.
These cartridges heating power is largely independent of the temperature they are at, but this is not the case with ceramic heaters.
For ceramic heaters the heating power is drastically reduced when the temperature increases.
I measured a ceramic heaters and found a start heating power of about 50 Watt at room temperature, but only about 24 Watt at 235 °C.
With MPC you need to provide the heater power, so I enter 24 Watt. This causes 2 issues:
1. MPC auto-tune might not provide accurate results if the actual heating power is not (relatively) constant.
2. Temperature overshoot when you set a hotend target temp lower than 235°C.
3. Slow response and it takes a long time to reach the hotend target temperature if the temperature is above 235°C.

### Additional context

I believe the common 6x20mm hotend heaters use nichrome wire inside. Nichrome has a very low temperature coefficient α = 0.0002 / 1 °C. This means that a temperature increase of 100 °C only increases the electrical resistance of by 2%.
I'm not sure what wire is used in ceramic heaters, but the temperature coefficient is much larger.
If the heater wire is a pure metal then the resistance should be linear (within reason) to the temperature, so only 2 points/measurents would be needed to calculate an estimation of the heater power at a given temperature.

ellensp, 2024-07-09:
MPC does not support ceramic heaters (ie PTC heaters)

see https://github.com/MarlinFirmware/Marlin/pull/23751#issuecomment-1229559530 for details

tombrazier, 2024-07-09:
Actually MPC works pretty well with PTC heaters even though it was not designed to. I have used it for ages with a TriangleLabs CHC.

What does not work well is the original asymptotic autotune. The differential autotune should work with PTC hotends but I haven't tested it. New newer default autotune behaviour is to try asymptotic and if that gives obviously bad results use differential. With PTC I would advise going straight for `M360 T S1` for differential.

Once tuned, I add a line like `M306 P30` to my slicer's filament specific gcode to tell MPC that the power at the target temperature is not the same as the power at room temperature.

A nice feature to add to MPC would be to be able to tell MPC that the hotend is PTC and give a coefficient so it can automatically calculate power from temperature and also so that it defaults straight to differential tuning.

rondlh, 2024-07-09:
> MPC does not support ceramic heaters (ie PTC heaters)
> 
> see [#23751 (comment)](https://github.com/MarlinFirmware/Marlin/pull/23751#issuecomment-1229559530) for details

Thanks for the clarification, I didn't know that, but from what I saw and measured I suspected this, so I raised this FR.

Practically I have been using MPC with PTC heaters for some time, and found it results in much more stable temperature behavior than PID. Internally I assume some calculation will be wrong and the autotune is likely to produce some incorrect values, but in the end it works fine as long as I provide a heater power that is close to reality at the working temperature (around 235°C).
E3D specifies the temperature coefficient of the Revo heaters, it seems adding this parameter together with the hotend heater voltage would help to improve the MPC model. This would also work for the common heater cartridges (6x20mm), but is not so important in that case.

**Revo heater temperature coefficient:**
https://e3d-beta.myshopify.com/pages/heatercore-range-datasheet
Initial Resistance of a 40W, 24V heater at 23°C: 14.4Ω
Initial Resistance of a 60W, 24V heater at 23°C: 9.5Ω
(Temp Coefficient of a 40W Heater Cartridge: 0.0004 °C^(-1)) NOT A PTC HEATER
Temp Coefficient of a 40W HeaterCore: 0.002078 °C^(-1)
Temp Coefficient of a 60W HeaterCore: 0.002078 °C^(-1)

The needed math is quite simple: Rt = R23 * (1+α * (t - 23))
Rt is the resistance at temperature t
R23 is the resistance at the reference temperature 23°C, the heater power is specified for this temperature.
α is the temperature coefficient.
V = I * R (Voltage = current * resistance)
P = I * V, with V stable at 12 or 24 Volt (Power = current * voltage)
Pt = (V * V) / Rt = V * V / (R23 * (1 + α * (t - 23)))
So if you know the temperature coefficient and hotend heater voltage then you can calculate the heater power dynamically.

For example:
For the Revo 40W PTC heater at 24V with a temperature coefficient of 002078/°C  we would find:
P23 = 40W = (V * V) / R = (24 * 24) / R23 ==> R23 = (24 * 24) / 40 = 14.4Ω (as specified).
P235 = (24 * 24) / (14.4 * (1 + 0.002078 * ( 235 - 23))) = 27.8 Watt at 235°C (as shown in the graph).

I use a 60 Watt Creality K1 style heater, the temperature coefficient is much bigger.
I found the heating power is 24 Watt at 235°C.
I will calculate the heater α based on this data:
P23 = 60 Watt = ( V * V) / R23 = (24 * 24) / R23 ==> R23 = 9.6Ω
P235 = 24 Watt (measured) = (24 * 24) / R235 ==> R235 = 24Ω
R235 = R23 * (1 + α * (235 - 23)) = 24Ω
24     = 9.6  * (1 + 212 * α)
24 / 9.6 = 1 + (212 * α)
α = ((24 /  9.6) - 1) / 212 = 0.007075/°C, this is about 3.5 times the Revo heater temperature coefficient.

tombrazier, 2024-07-10:
@rondlh If I code it will you test it?

rondlh, 2024-07-10:
> @rondlh If I code it will you test it?

I will, no problem!

tombrazier, 2025-06-09:
> > [@rondlh](https://github.com/rondlh) If I code it will you test it?
> 
> I will, no problem!

Yes still interested in testing this @rondlh?

closed 2025-06-14 by thinkyhead