MarlinFirmware/Marlin #27224: [BUG] Ender 5 Pro does not start with Marlin 2.1.2.4

mlutescu, 2024-06-30:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

In fact, I could not compile the bugfix-2.1.x branch.

The behaviour is exactly the one described in issue #22898 https://github.com/MarlinFirmware/Marlin/issues/22898:  the printer is looping between the "Ender" screen and the "Marlin" screen iff
 the SD Card is inserted.   If the SD Card is taken out then imediately the loop stops and the device boots correcly.   I suspect this might be a regression.  The configuration files are copied from _config/examples/Creality/Ender-5 Pro/CrealityV422_ in the  example configurations bundle.

[configs.zip](https://github.com/user-attachments/files/16043038/configs.zip)

### Bug Timeline

in release 2.1.2.4.    In release 2.1.2.3 it works correctly.

### Expected behavior

I expect the printer to boot with the SD Card inserted.

### Actual behavior

I have to take out the SD Card for the printer to boot

### Version of Marlin Firmware

2.1.2.4

### Printer model

Ender-5 Pro

### Electronics

Stock electronics

### LCD/Controller

does not matter

### Other add-ons

none

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-06-30:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

mlutescu, 2024-06-30:
I confirm the bug does NOT reproduce with branch _bugfix-2.1.x_.   In the initial report I mentioned I could not compile the code from branch  _bugfix-2.1.x_ but I did not realise I had to copy the config files for this branch too (I errorneosly  used the config files one from the 2.1.2.4 bundle).

ellensp, 2024-06-30:
There are several build environments for this controller, which of the following did you use? and what chip do you have so we can try and replicate? 
STM32F103RE_creality
STM32F103RE_creality_xfer
STM32F103RC_creality
STM32F103RC_creality_xfer
STM32F103RE_creality_maple

mlutescu, 2024-06-30:
@ellensp _STM32F103RE_creality_ is the environment and I don't know what is the chip of the controller.    Is any method to find the chip version without disassembling the printer?  (I would rather not do this.  As long as the bug does not reproduce on branch `bugfix-2.1.x`, this is good enough for me)

ellensp, 2024-06-30:
I just tried 2.1.2.4 on my STM32F103RET6 creality v4.2.2  
It booted fine, sdcard is still in the sdcard slot 
rebooted without sdcard, 
rebooted with scard in slot again, no issues

Korrel307, 2024-10-07:
@ellensp I am experiencing the same problem. Please let me know what information I can provide to help with your testing

closed 2025-01-22 by github-actions[bot]