MarlinFirmware/Marlin #27226: [FR] add or augment Spindle Laser function to use PWM driven external circuit

rflulling, 2024-07-01:
### Is your feature request related to a problem? Please describe.

**Is your feature request related to a problem?**
I have an external Spindle Driver for standard brushed type DC motor I would like to use.

- I need to be able to set M3 using the standard M3 S1 to M3 S50000, which is already part of the configuration as RPM. 
- However I need this to output as a PWM, with a configurable duty cycle. Default being assumed 50%.
- Additionally, for my purposes, Speed_Power_Min 1000, and Speed_Power_Max 3000, are as complicated as this needs to be. I wont ask for new variables. 
- While I am at it I would also also like to go a step further and request that the defined configurable pins, specific to a laser spindle setup be added to the laser spindle section. I see no reason they should be manually added to a board pins file. Unless we are going to require the same of all other locations in the conf where a pin is named to complete a define. 

### Are you looking for hardware support?

**Are you looking for hardware support?**No. This is NOT a request support for a controller or LCD where in code will be installed. This is not a request for assistance with wires or any physical setup configuration issue. This request is for Marlin FW in general but specific to Spindle and Laser section. I humbly request those who previously worked on the code for the laser CNC sections to make these changes. 

### Describe the feature you want

**Describe the feature you want:** 
To augment the existing code slightly to support connectivity of external circuits for use in Spindle Control. Long term changes applied may also benefit others who may want to use various stand alone external controllers for spindle or laser with Marlin. 

- Perhaps this might be accomplished simply by exposing duty cycle and leaving it at -1 for typical configuration and user configured for all applications where a specific range is needed. 
- I reason that why I am unable to use CUTTER_POWER_UNIT "RPM" is that the PWMDC which is not exposed. 

- Might I suggest SPEED_POWER_PWMDC to be nested within #if ENABLED(SPINDLE_LASER_USE_PWM), and then followed by "// -1 to disable, for default function, 1-100 to set duty cycle, higher numbers being more ON time"

- The circuit I am driving requires a 50% duty cycle, 1000hz PWM represents 1%, and 3000hz PWM 100%. It is a very simple circuit to offload actual load from control board mosfets which are not all able to handle the load of a spindle board to board. This one supports up to 12A. 

- Coincidentally, switching over to a dedicated stand alone external spindle controller. I tinkered with it's own PWM duty cycle. I had previously remarked on Discord how Marlin was vastly more noisy than the GRBL controller I swapped out, and now I think I understand why. The default internal value for PWMDC must be currently less than 16000, a guess will say it's actually more like 14840. Anything under 1600 starts to make this low quality Spindle vibrate badly. The default for the stand alone is 20000, with a range of 1k to 99k. It is probable I will need to set a different duty cycle when testing under load.  Perhaps this may also be a difference in spindle start up, where duty cycle may be on par with minimum default speed for spindle initialization. Gcode could be set to adjust duty cycles depending on load. -however this would be more valuable when a spindle is connected directly to a mosfet on the controller like say the heated bed. -where my tiny external is concerned, voltage output to the spindle is constant and the duty cycle only effects the onboard SSR. 

- Additionally exposing the lines to set the pins for Enable, Direction, and PWM should simplify set up for many others, I would like to hope. Plus depending on how this is all implemented open up a wider variety of common external drivers that might be used with spindle laser, ie @section CNC.

### Additional context

Suggested layout of the laser spindle section, for **reference** only. To draw attention to points suggested to be added within conf adv. 

```
#define SPINDLE_FEATURE
//#define LASER_FEATURE
#if ANY(SPINDLE_FEATURE, LASER_FEATURE)
  //#define SPINDLE_LASER_ACTIVE_STATE HIGH    // Set to "HIGH" if SPINDLE_LASER_ENA_PIN is active HIGH
  //#define SPINDLE_LASER_ENA_PIN PC5          // Enable Pin. Not included by default
  //#define SPINDLE_DIR_PIN PE0                // Direction Pin. Not included by default

  #define SPINDLE_LASER_USE_PWM                // Enable if your controller supports setting the speed/power
  #if ENABLED(SPINDLE_LASER_USE_PWM)
    #define SPINDLE_LASER_PWM_PIN PE1          // Not included by default
    #define SPINDLE_LASER_PWM_INVERT false     // Set to "true" if the speed/power goes up when you want it to go slower
    //#define SPINDLE_LASER_FREQUENCY 30000    // (Hz) Spindle/laser frequency (only on supported HALs: AVR, ESP32, and LPC)
                                               // ESP32: If SPINDLE_LASER_PWM_PIN is onboard then <=78125Hz. For I2S expander
                                               //  the frequency determines the PWM resolution. 2500Hz = 0-100, 977Hz = 0-255, ...
                                               //  (250000 / SPINDLE_LASER_FREQUENCY) = max value.
  #endif

  //#define AIR_EVACUATION                     // Cutter Vacuum / Laser Blower motor control with G-codes M10-M11
  #if ENABLED(AIR_EVACUATION)
    #define AIR_EVACUATION_ACTIVE       LOW    // Set to "HIGH" if the on/off function is active HIGH
    //#define AIR_EVACUATION_PIN        42     // Override the default Cutter Vacuum or Laser Blower pin
  #endif

  //#define AIR_ASSIST                         // Air Assist control with G-codes M8-M9
  #if ENABLED(AIR_ASSIST)
    #define AIR_ASSIST_ACTIVE           LOW    // Active state on air assist pin
    //#define AIR_ASSIST_PIN            44     // Override the default Air Assist pin
  #endif

  //#define SPINDLE_SERVO                      // A servo converting an angle to spindle power
  #ifdef SPINDLE_SERVO
    #define SPINDLE_SERVO_NR   0               // Index of servo used for spindle control
    #define SPINDLE_SERVO_MIN 10               // Minimum angle for servo spindle
  #endif

  /**
   * Speed / Power can be set ('M3 S') and displayed in terms of:
   *  - PWM255  (S0 - S255)
   *  - PERCENT (S0 - S100)
   *  - RPM     (S0 - S50000)  Best for use with a spindle
   *  - SERVO   (S0 - S180)
   */
  #define CUTTER_POWER_UNIT RPM

  /**
   * Relative Cutter Power
   * Normally, 'M3 O<power>' sets
   * OCR power is relative to the range SPEED_POWER_MIN...SPEED_POWER_MAX.
   * so input powers of 0...255 correspond to SPEED_POWER_MIN...SPEED_POWER_MAX
   * instead of normal range (0 to SPEED_POWER_MAX).
   * Best used with (e.g.) SuperPID router controller: S0 = 5,000 RPM and S255 = 30,000 RPM
   */
  //#define CUTTER_POWER_RELATIVE              // Set speed proportional to [SPEED_POWER_MIN...SPEED_POWER_MAX]

  #if ENABLED(SPINDLE_FEATURE)
    //#define SPINDLE_CHANGE_DIR               // Enable if your spindle controller can change spindle direction
    #define SPINDLE_CHANGE_DIR_STOP            // Enable if the spindle should stop before changing spin direction
    #define SPINDLE_INVERT_DIR          false  // Set to "true" if the spin direction is reversed

    #define SPINDLE_LASER_POWERUP_DELAY   5000 // (ms) Delay to allow the spindle/laser to come up to speed/power
    #define SPINDLE_LASER_POWERDOWN_DELAY 5000 // (ms) Delay to allow the spindle to stop

    /**
     * M3/M4 Power Equation
     *
     * Each tool uses different value ranges for speed / power control.
     * These parameters are used to convert between tool power units and PWM.
     *
     * Speed/Power = (PWMDC / 255 * 100 - SPEED_POWER_INTERCEPT) / SPEED_POWER_SLOPE
     * PWMDC = (spdpwr - SPEED_POWER_MIN) / (SPEED_POWER_MAX - SPEED_POWER_MIN) / SPEED_POWER_SLOPE
     */
    #if ENABLED(SPINDLE_LASER_USE_PWM)
      #define SPEED_POWER_PWMDC        50    // -1 to disable, 50 default, high values are higher on time
      #define SPEED_POWER_INTERCEPT      0     // (%) 0-100 i.e., Minimum power percentage
      #define SPEED_POWER_MIN         1000     // (RPM)
      #define SPEED_POWER_MAX         3000     // (RPM) SuperPID router controller 0 - 30,000 RPM
      #define SPEED_POWER_STARTUP     1000     // (RPM) M3/M4 speed/power default (with no arguments)
    #endif

  #else
```

ellensp, 2024-07-01:
"I see no reason they should be manually added to a board pins file." you don't need to do this.

Just because the pin define is not In the config files does not mean you cannot add it. It doesn't need to be there by default to enable it....  Just add it the line.

rflulling, 2024-08-01:
Just checking in to see if there were any ideas, or suggestions.

Bob-the-Kuhn, 2024-11-29:
I'm confused as to what you need to drive the speed input on your RPM controller.  Please provide a model number or pointer to the RPM controller so I can verify the requirements.

The current Marlin firmware assumes an analog out is used to control the RPM and it has a linear transfer function.  The analog out function is almost always implemented by using a PWM output.  The PWM signal itself is a fixed frequency pulse train whose pulse width is varied to set an average voltage output.  The one RPM controller I know ([SuperPID](https://vhipe.com/)) assumes this setup is driving it's speed input pin.

You mentioned that your RPM controller needs a variable frequency to control it.  That function isn't implimented in Marlin (as far as I know).  I expect it can be but the low level drivers would need to be created for each processor type.