MarlinFirmware/Marlin #27227: [FR] Adding PID bed tuning

kclambi1, 2024-07-01:
### Are you looking for hardware support?

Neptune 3 Pro

### Describe the feature you want

Hi!  Wonderful firmware so far; seen no quality diminishing in my tuned prints quality and the bed leveling improvements/speed are noticeable.  No bugs so far, but will keep you posted!

What would be nice is additional implementation of NARUTO's PID bed firmware tweaks to go along with the nozzle PID calibration since Elegoo decided to comment that out and go with bang-bang to keep the LED lights from flickering.  I have no problem with the flicker.  Other than that, I think everything useful has already been incorporated by you already, so thanks again!  :)

thisiskeithb, 2024-07-01:
If you want to enable PID on the bed, enable `PIDBEDTEMP`: https://github.com/MarlinFirmware/Marlin/blob/c81416438d5a4779b1bdb103e9a166252b39ae96/Marlin/Configuration.h#L762-L774

closed 2024-07-01 by thisiskeithb

kclambi1, 2024-07-01:
Thank you thisiskeithb!  This was totally supposed to go on a custom Marlin firmware github and not the main; DOH!  I can tell it's a Monday!  Thank you!!