MarlinFirmware/Marlin #27228: lvgl ui (mks ui) m600 filament change [BUG] (bug summary)

Luisdjnas, 2024-07-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The M600 filament change function is not operational in the lvgl ui (mks ui) interface.
Lvgl ui (mks ui) interface does not show PID menu

### Version of Marlin Firmware

2.1.2.4.

### Printer model

Prusa I3

### Electronics

MKS Robin nano v3.1

### LCD/Controller

TS35

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-07-01:
Please zip up and attach your configs here. I cannot download or copy the files from your dropbox without an account.

Luisdjnas, 2024-07-02:
Vale, no sabía cómo subir los archivos. Los subiré a WeTransfer también pero se los adjunto aquí.

Luisdjnas, 2024-07-02:
Sorry, I didn't realize you can attach a .zip, I've already uploaded it.

Luisdjnas, 2024-07-02:
A friend told me about making some changes to solve the compilation errors but it still does not appear in the menu and when I execute the M600 command the printer starts the filament change process but I cannot give it to continue inserting the filament later.

- at line 882 of file \Marlin\src\inc\Conditionals_adv.h,
add HAS_TFT_LVGL_UI

#if ANY(EXTENSIBLE_UI, IS_NEWPANEL, EMERGENCY_PARSER, HAS_ADC_BUTTONS, HAS_DWIN_E3V2, HAS_TFT_LVGL_UI)

- in line 2933 of file \Marlin\src\inc\Conditionals_post.h, add HAS_TFT_LVGL_UI

#if ANY(HAS_MARLINUI_MENU, EXTENSIBLE_UI, DWIN_LCD_PROUI, DWIN_CREALITY_LCD_JYERSUI, HAS_TFT_LVGL_UI) || ALL(EMERGENCY_PARSER, HOST_PROMPT_SUPPORT)

With these options it is not necessary to activate EMERGENCY_PARSER

Luisdjnas, 2024-07-06:
I still have the problem, I think it is because the ts35 v2.0 device already has its own parking position.
Would there be a way to pause the printing using gcode as if you hit the pause button on the screen?
That would be enough to be able to change the filament.

Luisdjnas, 2024-07-16:
I have seen that it does not show the PID menu on the screen either.  And you cannot define pause.

thisiskeithb, 2024-07-16:
> I still have the problem, I think it is because the ts35 v2.0 device already has its own parking position.

No, the TS35 doesn't run separate firmware. The parking position is hard coded in LVGL UI.

> I have seen that it does not show the PID menu on the screen either. And you cannot define pause.

Keep in mind that MKS's LVGL UI is a custom UI that does not use many of Marlin's standard methods or config settings. MKS also no longer maintains the UI, so it's on us/the community to fix up various shortcomings.

I started on some of these improvements, but it's **_a lot_** of work.

I recommend using `TFT_CLASSIC_UI` or `TFT_COLOR_UI` for now since those UIs are well-supported.

Luisdjnas, 2024-07-16:
Hello, thank you very much for your dedication, if you could help correct errors in the interface I would be delighted.
I hope we can fix the LVGL interface.
I am currently working with the TFT_COLOR_UI interface in which I have modified with help something that I share with you to be able to implement it. The remaining printing time and percentage were not marked.
I have modified 

\Marlin\src\lcd\tft\ui_480x320.cpp

y += TERN(HAS_UI_480x272, 36, 44);
  // print duration
  char buffer1[14];
  char buffer2[14];
  duration_t elapsed = print_job_timer.duration();
  duration_t resto = ui.get_remaining_time();
  elapsed.toDigital(buffer1);
  resto.toDigital(buffer2);

  const uint8_t porcentaje = ui.get_progress_percent();
  tft.canvas((TFT_WIDTH - 300) / 2, y, 300, 29);
  tft.set_background(COLOR_BACKGROUND);
  tft_string.set("|<: ");
  tft_string.add(buffer1);
  tft_string.add("  ");
  tft_string.add(pcttostrpctrj(porcentaje));
  tft_string.add("  >|: ");
  tft_string.add(buffer2);
  tft.add_text(tft_string.center(300), 0, COLOR_PRINT_TIME, tft_string);

I upload the modified file.

[ui_480x320.zip](https://github.com/user-attachments/files/16254989/ui_480x320.zip)

Luisdjnas, 2024-07-16:
This is the final look.
[image]

Luisdjnas, 2024-10-15:
I still have the same problem and can't solve it.

closed 2025-01-30 by github-actions[bot]