MarlinFirmware/Marlin #27236: [BUG] Potential bug in M501 logic with version mismatch

rondlh, 2024-07-02:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I changed some settings in my firmware which causes the EEPROM size to change, but it's reported as a version number errors twice, which is odd.

```
echo:EEPROM version mismatch (EEPROM=? Marlin=V90)
echo:EEPROM version mismatch (EEPROM=? Marlin=V90)

```
I have EEPROM_SETTINGS  enabled, and also EEPROM_AUTO_INIT to initilize the factory defaults, but this does not work.
To my surprise if found that lots of settings are 0:

```
M306
M306 E0 P0.00 C0.00 R0.0000 A0.0000 F0.0000 H0.0000

```

I believe there is an error in the logic of M501 in settings.cpp

```
  /**
   * M501 - Retrieve Configuration
   */
  EEPROM_Error MarlinSettings::_load() {
    EEPROM_Error eeprom_error = ERR_EEPROM_NOERR;

    const EEPROM_Error check = check_version();
    if (check == ERR_EEPROM_VERSION) return eeprom_error;
```

This is odd because if check_version() returns a "version error", then _load() returns "no error".

I think it should be something like this (difference in the last line):

```
  /**
   * M501 - Retrieve Configuration
   */
  EEPROM_Error MarlinSettings::_load() {
    EEPROM_Error eeprom_error = ERR_EEPROM_NOERR;

    const EEPROM_Error check = check_version();
    if (check == ERR_EEPROM_VERSION) return ERR_EEPROM_VERSION; 
```

The double version mismatch error can probably be explained by the code below. validate() calls _load(), and if no validation error is detected then _load() is called again. But actually _load() seems to return the wrong status.

```
  bool MarlinSettings::load() {
    if (validate()) {
      const EEPROM_Error err = _load();
```

```
  bool MarlinSettings::validate() { // IRON, TRUE IF OK, FALSE IS ERROR
    validating = true;
    #ifdef ARCHIM2_SPI_FLASH_EEPROM_BACKUP_SIZE
      EEPROM_Error err = _load();
      if (err != ERR_EEPROM_NOERR && restoreEEPROM()) {
        SERIAL_ECHOLNPGM("Recovered backup EEPROM settings from SPI Flash");
        err = _load();
      }
    #else
      const EEPROM_Error err = _load();
    #endif
    validating = false;

    if (err) ui.eeprom_alert(err);

    return (err == ERR_EEPROM_NOERR);
  }
```

### Bug Timeline

No sure, in the last 6 months I guess

### Expected behavior

Reset settings to factory defaults when and EEPROM error is detected.

### Actual behavior

Factory defaults are not loaded, a double errors message is given.

### Steps to Reproduce

1. Create a corrupt EEPROM storage, for example by changing the number of bed probe points.
2. Update to the new firmware and check for the double error message and the loaded defaults like PID/MPC settings
3. In the pin-file define FLASH_EEPROM_EMULATION (I cannot use the onboard EEPROM because I use the CAN bus).
and uncomment: I2C_EEPROM

### Version of Marlin Firmware

Bugfix 2.1.x June 30st 2024

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### LCD/Controller

BTT TFT35 V3.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config and Pins.zip](https://github.com/user-attachments/files/16074681/Config.and.Pins.zip)

dbuezas, 2024-07-09:
I think it used to return ERR_EEPROM_NOERR instead.

ellensp, 2024-07-09:
it used to be "if (!EEPROM_START(EEPROM_OFFSET)) return eeprom_error;"

see

https://github.com/MarlinFirmware/Marlin/pull/27199/files#diff-3b2c0da2300fa99c611618bb2d1a9f7e81edb092aac5d49c83a6d9068561838cL1806

dbuezas, 2024-07-09:
Look at the line above. `eeprom_error = ERR_EEPROM_NOERR`

closed 2024-07-10 by thinkyhead