MarlinFirmware/Marlin #27237: mks wifi does not read or print files via cura

odenibsb, 2024-07-03:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Since version 2.2.3 I can no longer read or start printing via cura. I'm using a mks wifi with mks robin nano. I can record via wifi through Cura but I can't read it after that. I downgraded to 2.2.1 and it works perfectly. I appreciate any possible help.

### Bug Timeline

since 2.2.3

### Expected behavior

Upload the files via WiFi and start printing.

### Version of Marlin Firmware

marlin 2.2.4

### Printer model

home made delta 

### Electronics

mks robin nano mks wifi 

### LCD/Controller

mks ts35  lvgl

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/16088833/Marlin.zip)

ellensp, 2024-07-03:
Config files do not match description, description says "mks robin nano" but the provided config says BOARD_MKS_EAGLE

odenibsb, 2024-07-03:
Yes, it's a MKS Eagle, both are the same board. the mks robin nano is better known...

thisiskeithb, 2024-07-03:
I've corrected the description.

thisiskeithb, 2024-07-03:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

odenibsb, 2024-07-03:
ok, sorry.

odenibsb, 2024-07-03:
> Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

ok, i do it.

odenibsb, 2024-07-04:
> > Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.
> 
> ok, i do it.

I tried using bugfix-2.1.x but I had no success with the files. Furthermore, another problem also appeared: I have no movements in homming and the system freezes.

VampireCrab83, 2024-07-05:
same problem here

odenibsb, 2024-07-05:
I reviewed the entire configuration using bugfix-2.1.x line by line and was unable to get Cura to read the files.

ellensp, 2024-07-05:
the changes from 2.1.2.2 and 2.1.2.3 are minimal..  https://github.com/MarlinFirmware/Marlin/compare/2.1.2.2..2.1.2.3

odenibsb, 2024-07-07:
right, it's version 2.1.2.1, I've already corrected it

closed 2024-10-22 by github-actions[bot]