MarlinFirmware/Marlin #27239: [BUG]G29 don't work for Ender3 v4.2.7 board

kinlougha, 2024-07-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The G29 command for bed leveling won't work in release 2.1.2.2, 2.1.2.3 and 2.1.2.4.
My setup was Ender3 with Creality v4.2.7 board with a BLToutch sensor and config BILINEARLEVELING.
When I issue a G29 gcode, the probe go to the first position (ie: 25,25) , go down slowly, toutch the bed and stop.
We have no error message on the Ender3 display but on serial terminal (pronterface connect on com port) I have this message: 

"Error:Probing Failed"

Everything work fine in 2.1.1.1. but I have the same behavier in 2.1.2.2 true 2.1.2.4

### Bug Timeline

New

### Expected behavior

A working g29 gcode

### Actual behavior

Print stop on G29 gcode

### Steps to Reproduce

Issue a G29 gcode in gcode file ou on serial terminal ou on the command pannel.
[Configuration_File_2.1.1.1.zip](https://github.com/user-attachments/files/16100884/Configuration_File_2.1.1.1.zip)
[Configuration_File_2.1.2.2.zip](https://github.com/user-attachments/files/16100887/Configuration_File_2.1.2.2.zip)
[Configuration_File_2.1.2.3.zip](https://github.com/user-attachments/files/16100894/Configuration_File_2.1.2.3.zip)
[Configuration_File_2.1.2.4.zip](https://github.com/user-attachments/files/16100897/Configuration_File_2.1.2.4.zip)

### Version of Marlin Firmware

2.1.2.2 - 2.1.2.3 - 2.1.2.4

### Printer model

Creality Ender-3

### Electronics

Creality v4.2.7

### LCD/Controller

Stock 12864 Ender LCD

### Other add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2024-07-04:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information:

- Download Marlin [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code. You will need to port your config/settings to this version.
- Enable `DEBUG_LEVELING_FEATURE` and `M114_DETAIL` and re-flash the firmware.
- Connect to your printer from host software such as [Pronterface](https://github.com/kliment/Printrun/releases/latest), [Repetier Host](https://www.repetier.com/download-now/), or [OctoPrint](https://octoprint.org/download/).
- Send `M502` and `M500` to ensure your Configurations are applied.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

kinlougha, 2024-07-04:
Hello, I have a problem with bugfix-2.1.x , I can't config my probe offset, it was at -45,-10,-4.65 but at compilation I have this message: 
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\HAL.cpp:26:
Marlin\src\HAL\STM32\../../inc/../core/macros.h:243:41: error: static assertion failed: NOZZLE_TO_PROBE_OFFSET.x must be within -20 and 235

If I put my offset at -20, -10, -4.65, on the g92 the probe is outside of the bed.

kinlougha, 2024-07-04:
OK I set to -2 by error.

 

Sorry.

board (Issue #27239)

 

What is PROBE_OFFSET_XMIN set to?


<https://github.com/MarlinFirmware/Marlin/issues/27239#issuecomment-22094205
WFQRAVCNFSM6AAAAABKLZBWOCVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDEMBZGQZDA
NJTGU> .
<https://github.com/notifications/beacon/AMS2ZCFXV22I7KUI7KL6WKTZKWFQRA5CNFS
M6AAAAABKLZBWOCWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTU

kinlougha, 2024-07-04:
Hello Keith, 
[LOG_bugfix2.1.x.txt](https://github.com/user-attachments/files/16102819/LOG_bugfix2.1.x.txt)
OK, change PROBE_OFFSET_XMIN to -50 and run Bugfix-2.1.x successfuly many times. See attach log file.

What next?

Thanks.

kinlougha, 2024-07-04:
Hello Keith,

I found the error. Il was a misunderstood of a config parm:

#define Z_PROBE_LOW_POINT          -4 
but my offset was -4.65 so the #define Z_PROBE_LOW_POINT  was to short for my offset. I put it at -5 and everything is OK.

Thanks for your help.

Alain K.

mriscoc, 2024-07-11:
Your Z offset is very large, check the recommended position of the BLTouch in the image:

[image]

So, a Z offset should be approximately in the range of -0.7 to -2.7

kinlougha, 2024-07-12:
I know that.

board (Issue #27239)

 

image.png (view on web)
<https://github.com/MarlinFirmware/Marlin/assets/2745567/08832e6c-28d0-4579-
bd6a-a5273ad429f4> 

<https://github.com/MarlinFirmware/Marlin/issues/27239#issuecomment-22232699
2RHDAVCNFSM6AAAAABKLZBWOCVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDEMRTGI3DS
OJZGI> .
<https://github.com/notifications/beacon/AMS2ZCFTBVSIV5QGBRQAK33ZL2RHDA5CNFS

closed 2024-10-25 by github-actions[bot]