MarlinFirmware/Marlin #27244: [FR] Blind PWM Charging a power loss recovery capacitor via a transistor

van0014, 2024-07-05:
### Is your feature request related to a problem? Please describe.

There is a bit of information on the Trigorilla Pro board, and it seems practical to implement this. There's a nice pin debug system in Marlin, and this was enough for me to do some tests with. I just couldn't implement it.

I would try to do it myself, but am not experienced enough with C. Only with Arduino, which isn't as advanced.

### Are you looking for hardware support?

Anycubic Delta - Trigorilla Pro, STM32

### Describe the feature you want

The ability to charge the onboard supercapacitor, by slowly ramping up power to it and maybe tapering off. This could be triggered when starting an SD card print, and possibly also through the debug pin state system

It might be possible to monitor the comparator to know when it's charged, otherwise the onboard 1F capacitor would have a predictable safe charging rate, and therefore can be charged without feedback.

### Additional context

There is some schematics on thingyverse, as well as pin function explainations.

https://www.thingiverse.com/thing:4271636

Strategy:
    -Slowly ramp up power going into the cap charge Pin PG4 at boot or SD card print start. Finish at 100% PWM
    -Monitor the comparator on Pin PG2 (#98). Pin seems to be LOW normally, and HIGH when 24v power is off
    Note: 1F capacitor lasts approx. 600-800ms, where it's enough to light an onboard LED. Voltage unknown at this point.

rondlh, 2024-07-11:
I don't fully understand, what would be the benefit of this feature?
Don't these backup power modules have a power fail signal/pin, that is used by Marlin to handle some last minute activities?

van0014, 2024-07-11:
Yes, they have a pin to signal power loss. What they don't have is a capacitor tied directly to the 5v rail. They are MOSFET driven, so the hardware incorporates charging of the capacitor. And if it doesn't get charged, it won't be able to stay on for long enough to trigger power loss actions. There's also an uncertain relationship between the charge pin and the power loss comparator; the charging pin is tied together with the non inverting or positive pin of the power loss pin comparator. I've done some basic tests with Marlin's debug feature. Unfortunately because I changed my stepper motors, I cannot easily run stock firmware to properly test how it functions when charging and during power loss