MarlinFirmware/Marlin #27245: [BUG] CR10S screen goes blank after updating to 2.x bugfix

manoman48, 2024-07-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When updating to latest bugfix version from 1.1.6 on CR10S with Mega2560 v2.2 silent board the firmware uploads successfully but on power cycle the screen turns on and is blank. Uploading original 1.1.6 version fixes this. Is there a big in what LCD screen type has been set up?

### Bug Timeline

New - on update to 2.1.bugfix

### Expected behavior

I expected it to have a working display

### Actual behavior

The display was blank

### Steps to Reproduce

Used VSCode to and Auto Marlin Build to compile firmware and upload.

### Version of Marlin Firmware

1.1.6

### Printer model

CR10S

### Electronics

v2.2 silent board

### Bed Leveling

No Bed Leveling

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/user-attachments/files/16114072/Marlin.zip)

ellensp, 2024-07-06:
Provided configs do not even compile 
You  get the following 
`static assertion failed: PREHEAT_2_TEMP_HOTEND (250) must be less than HEATER_0_MAXTEMP (250) - 15.
`
The line "#define PREHEAT_2_TEMP_HOTEND 250" is not valid. The max temperature setting is 
"#define PREHEAT_2_TEMP_HOTEND 235" when HEATER_0_MAXTEMP is set to 250 and HOTEND_OVERSHOOT is set to 15

ellensp, 2024-07-06:
What lcd type do you have?
Take a look here https://marlinfw.org/docs/hardware/controllers.html#cr10_stockdisplay is this your display?

If on the lcd your using exp1 and exp2 (ie two cables) this is REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER 
If on the lcd your using exp3 (ie just the one cable) this is CR10_STOCKDISPLAY

manoman48, 2024-07-06:
> What lcd type do you have? Take a look here https://marlinfw.org/docs/hardware/controllers.html#cr10_stockdisplay at the CR10_STOCKDISPLAY is this your display?
> 
> If on the lcd your using exp1 and exp2 (ie two cables) this is REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER If on the lcd your using exp3 (ie just the one cable) this is CR10_STOCKDISPLAY

manoman48, 2024-07-06:
Thanks for replying. It uses two cables. It's a CR10S model. I'm just using the example config files for that model and I'm sure before they just worked without any changes.

ellensp, 2024-07-06:
I don't have a  creality v2.2, but a ramps 1.4 its close enough.  (all the same pin on the exp1 and exp2 ports)
Running your configs on a stock ramps + a CR10_STOCKDISPLAY in two wire REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER mode,

Works fine,  I cannot replicate this issue 

Are you building with vscode , as you should be.

manoman48, 2024-07-06:
Hi. Thank you. That got the screen working but all the rest of the configuration is wrong so I'm doing something stupid with the files and build process in vscode. It's been 18 months so I'm obviously rusty.

I'm going to start again over the next couple of days and go back through some vscode walkthroughs. Will report back on what happens.

Thank you again for responding. At least I know I can still print even if it's on the terrible Creality firmware. The default homing movements are brutal!

closed 2024-07-06 by ellensp