MarlinFirmware/Marlin #27246: DELAY_BEFORE_PROBING not working[BUG] (bug summary)

vqh2308, 2024-07-06:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

I am using a BTT Microprobe for probing. The type of this probe is FIX_MOUNTED_PROBE in configuration file. When z homing, the system does not give the probe time to deploy but activate immediately. Therefore, the probe trigger the first time immediately after the bed lower 10mm and cause an error. When I deploy the probe before z homing, the error does not appear. When I increase the value of DELAY_BEFORE_PROBING to 2000 ms, nothing change. The motherboard is BTT Octopus V1.1

### Bug Timeline

The issue started from v2.1.2.3 but still exist in v2.1.2.4

### Version of Marlin Firmware

v2.1.2.4

### Electronics

BTT Microprobe, BTT Octopus

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/user-attachments/files/16114715/Configuration.zip)

thisiskeithb, 2024-07-06:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

vqh2308, 2024-07-07:
Thank you for the reply. After I tried the newest Marlin bugfix-2.1.x, other problems occur. The probe behave unusual when the machine start. It only deploys to open state, not goes up and down as it should be. Also, I cannot control any motor with the bugfix firmware, so I cannot home to test how the probe work.

The file below is my configuration. 

[Marlin.zip](https://github.com/user-attachments/files/16118018/Marlin.zip)

dbuezas, 2024-07-07:
FYI: I have a working Microprobe v2 (same as you) running on latest bugfix-2.1.x on an skr3.

> It only deploys to open state, not goes up and down as it should be 

Does it not deploy and retract 3 times when powered up? This is not controlled by marlin, the probe does this by itself so if that doesn't work then the microprobe may be damaged. I confirmed this by turning the power on while holding the board's reset button (which prevents the MCU to start executing code)

vqh2308, 2024-07-08:
Yes, the probe does not deploy and retract 3 times as it should be. The problem only occur when I run marlin bugfix version. The older version works fine. However, I have a bigger problem is that none of my stepper motor driver works with the bugfix version. I don't know if there is anything wrong with it.

thisiskeithb, 2024-07-09:
> Yes, the probe does not deploy and retract 3 times as it should be. 

If your probe does not reliably self-test on power on, then it's a hardware issue with the probe. Marlin does not control this.