MarlinFirmware/Marlin #27251: [BUG] XYZ Min Endstop not working, even if the endstop trigged

AnandRaj-R, 2024-07-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

XYZ Min Endstop not working, even if the endstop trigged

### Expected behavior

Stepper motor must stop rotating, after hitting/triggering endstop switch/sensor

### Actual behavior

Stepper motor contineously rotating, after hitting/triggering endstop switch/sensor

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Anet A8

### Electronics

BTT Skr2

### LCD/Controller

REPRAP_DISCOUNT_SMART_CONTROLLER

### Other add-ons

Optical Endstop for X and Y axis

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_config.zip](https://github.com/user-attachments/files/16133084/Marlin_config.zip)

ellensp, 2024-07-08:
Endstops by default are only checked on homing, not on moves, unless you enable ENDSTOPS_ALWAYS_ON_DEFAULT

closed 2024-07-09 by thisiskeithb

AnandRaj-R, 2024-07-09:
I think your are not understanding what i mentioned, the actual issue is XYZ homing not working. (when i press Auto homing X and Y axis going to home location(X min & Y min) and hitting the endstop and crashing.

Missing endstop Min/Max definition, i think this is the cause for my homing problem 

#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG

ellensp, 2024-07-09:
Those defines are not required anymore, so where deliberately removed from bugfix

ellensp, 2024-07-09:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

AnandRaj-R, 2024-07-09:
How my printer knows the endstop location, that is where the endstops are mounted min or Max.

ellensp, 2024-07-09:
from:
#define X_HOME_DIR -1
#define Y_HOME_DIR -1
#define Z_HOME_DIR -1

your homing to min, so endstop is plugged into min

AnandRaj-R, 2024-07-09:
Thank you, 

> from: #define X_HOME_DIR -1 #define Y_HOME_DIR -1 #define Z_HOME_DIR -1
> 
> your homing to min, so endstop is plugged into min

Thank you, i noob, i will check 
Thank you for all.

AnandRaj-R, 2024-07-09:
> from: #define X_HOME_DIR -1 #define Y_HOME_DIR -1 #define Z_HOME_DIR -1
> 
> your homing to min, so endstop is plugged into min

By default all endstops are defined as -1=Min location, even if why homing not working.