MarlinFirmware/Marlin #27252: [BUG] PR #27249 breaks build (Visual Code, Windows)

vovodroid, 2024-07-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

PR https://github.com/MarlinFirmware/Marlin/pull/27249 breaks build (Visual Code, Windows)

```
PS C:\DISK_D\3D\Marlin> platformio run -e mega1280
Processing mega1280 (board: megaatmega1280; platform: atmelavr@~4.0.1; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
UnicodeDecodeError: 'charmap' codec can't decode byte 0x90 in position 96864: character maps to <undefined>:
  File "C:\Users\QQQ\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 167:
    env.SConscript(env.GetExtraScripts("pre"), exports="env")
  File "C:\Users\QQQ\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 612:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\QQQ\.platformio\packages\tool-scons\scons-local-4.7.0\SCons\Script\SConscript.py", line 279:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\DISK_D\3D\Marlin\buildroot\share\PlatformIO\scripts\preflight-checks.py", line 161:
    sanity_check_target()
  File "C:\DISK_D\3D\Marlin\buildroot\share\PlatformIO\scripts\preflight-checks.py", line 151:
    text = file.read()
  File "C:\Users\QQQ\.platformio\python3\Lib\encodings\cp1252.py", line 23:
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
```

### Bug Timeline

PR #27249

### Version of Marlin Firmware

Latest bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2024-07-09:
Please attach your configuration files.

vovodroid, 2024-07-09:
> Please attach your configuration files.

Just default in bugfix-2.1.x branch - https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.1.x/Marlin

 My own build fails as well.

ellensp, 2024-07-09:
confirmed, oddly this doesn't happen on linux at all with same code and config

ellensp, 2024-07-09:
Fixed. https://github.com/MarlinFirmware/Marlin/commit/8b81aae8704512d6c77f09e991eb77c800654d2e

closed 2024-07-09 by ellensp

ellensp, 2024-07-09:
If anyone's interested

it breaks on the Æ in Configuration_adv.h from the line
      //#define TOUCH_UI_UTF8_SCANDINAVIAN  // Æ Ð Ø Þ æ ð ø þ