MarlinFirmware/Marlin #27253: [BUG] Wrong MPC settings can cause thermal runaway

rondlh, 2024-07-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I already reported this in #27223, but no response there and I think this issue needs attention.

Because of #27223 all my MPC settings were set to 0, which causes a thermal runaway of the hotend, the heater keeps heating 100% while the target temp is not changes (still set to 0). I use a ceramic heater which did not survive this, even though I have thermal protection enabled.

### Bug Timeline

Introduction of MPC (I guess)

### Expected behavior

MPC doesn't allow bad values, to prevent a thermal runaway.

### Actual behavior

Hotend thermal runaway (broken ceramic heater)

### Steps to Reproduce

1. Set all MPC parameters to 0, and save them with M500
2. Restart the printer and monitor the hotend temperature
3. Switch the printer off asap when you see the heater start heating or unplug the hotend heater
RECOVERY: Run M502, M500 and restart the printer

### Version of Marlin Firmware

Latest bugfix July 2024

### Printer model

Custom

### Electronics

MKS Monster8 V1.0

### LCD/Controller

BTT TFT35 V3.0

### Other add-ons

BTT EBB42 V1.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I suggest to set some limits to the MPC parameters where needed.
I guess all values should be positive, and MPC_HEATER_POWER should have a minimum value (10?).
Setting some values to 0 might cause a divide by 0 issue, causing unpredictable results.

[Configuration_adv.zip](https://github.com/user-attachments/files/16140443/Configuration_adv.zip)

thisiskeithb, 2024-07-09:
@tombrazier Can you take a look at this?

tombrazier, 2024-07-09:
Protecting against dangerous bad config is a very good idea.

I am concerned that the values all ended up as zero in the first place, though. Did this happen with `M306 T`?

I think the following is needed:
- The config that most needs protecting against obviously wrong values is the heater power and the block heat capacity.
- Add sanity checks for compiled-in config.
- Add checks in M306.cpp.
- `M306 T` should not create unsafe config.

ellensp, 2024-07-09:
0 is result of bug https://github.com/MarlinFirmware/Marlin/issues/27236

rondlh, 2024-07-10:
> Protecting against dangerous bad config is a very good idea.
> 
> I am concerned that the values all ended up as zero in the first place, though. Did this happen with `M306 T`?

No, there is an issue in the loading of the settings, I reported it here: #27253

> I think the following is needed:
>     * The config that most needs protecting against obviously wrong values is the heater power and the block heat capacity.
>     * Add sanity checks for compiled-in config.
>     * Add checks in M306.cpp.
>     * `M306 T` should not create unsafe config.

That makes perfect sense to me. There is a report that somebody got negative values, that seems quite odd: #27181 I see you are onto that one already. Also make sure no divide by zero can occure.

thisiskeithb, 2024-07-11:
> No, there is an issue in the loading of the settings, I reported it here: https://github.com/MarlinFirmware/Marlin/issues/27236.

*was* an issue. That's now patched 🙂

rondlh, 2024-07-11:
> > No, there **is an issue** in the loading of the settings, I reported it here: #27236.
> 
> _was_ an issue. That's now patched 🙂

Right, tested it, works fine now, great job🙂

thijstriemstra, 2024-07-11:
guess it can be closed then.

thisiskeithb, 2024-07-11:
> guess it can be closed then.

#27236 is already closed. We still need to install some guardrails around sane MPC settings, so this (#27253) bug report can remain open.

thisiskeithb, 2024-08-06:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/27274

closed 2024-08-06 by thisiskeithb