MarlinFirmware/Marlin #27256: Ender 2 Pro Halted when homing

AcrZg, 2024-07-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi after build my Marlin 2.1.X FW and loaded to my ender 2 Pro, the printer boots perfect, but when tries to move or go home reports the homing fault and says Printer Halted

Motor Driver is MS35775 and I have it configured as  TMC2209_STANDALONE but also tried TMC2208_STANDALONE and A4988 with the same result

### Expected behavior

it is suposed to move to home and hit the endstops limit switch

### Actual behavior

is not moving at all :(

### Version of Marlin Firmware

Bugfix 2.1.x

### Printer model

Creality Ender 2 Pro

### Electronics

Stock electronics with MB 2.4 S4, motor drivers MS35775

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin_config.zip](https://github.com/user-attachments/files/16152500/Marlin_config.zip)

AcrZg, 2024-07-10:
I forgot to mention that I´m using the Ender2 Pro config examples from the Marlin 2.1.X Configs repo :)

ellensp, 2024-07-10:
MS35775 is TMC2208_STANDALONE

If you downloaded Marlin bugfix in the last 2 weeks, get the current version. there was a bug where default settings where not setup correctly in the eeprom. 

you should also enable at least one of these

  //#define EEPROM_AUTO_INIT  // Init EEPROM automatically on any errors.
  //#define EEPROM_INIT_NOW   // Init EEPROM on first boot after a new build.

AcrZg, 2024-07-10:
Hi thanks, I downloaded the bugfix monday night and I just saw that there has been changes yesterday night, I downloaded and tested and is working now :) thanks